balena-io/etcher #2944: Etcher should prompt the user to get the permissions to Full Disk access in MacOS Catalina

mangelino, 2019-10-20:
- **Etcher version:** 
1.5.59
- **Operating system and architecture:** 
MacOs Catalina
- **Image flashed:** 
All
- **Do you see any meaningful error information in the DevTools?** 

Etcher does not work in MacOS Catalina due to the new permission model. Adding Etcher to **Settings --> Security & Privacy --> Privacy --> Full Disk Access** fixes the issue.
Etcher should check this on startup and prompt the user to add the permission.

lurch, 2019-10-20:
For issues with Etcher on macOS Catalina, see #2833

thundron, 2019-10-22:
Thank you for reporting this along with the temporary workaround. We have some others written in #2833 and #2911 (mainly, run as sudo) and we're working on the PR to notarize the app correctly.
Closing as duplicate

closed 2019-10-22 by thundron