balena-io/etcher #2945: Etcher - USB Drive Inaccessible

syedsameerali, 2019-10-20:
I was facing some difficulty with a USB thumb drive, can no longer be accessed on Windows or mounted in a Linux environment.

I am reporting this issue since I have noticed similar issues opened by other etcher users facing the same outcome for their USB thumb drives.

I am not certain, but it is possible that etcher contributed to the failure of the device. I could be mistaken as there were other steps involved that led up to the usb's drives failure.

After flashing a Raspbian Buster image to the USB in question with Etcher. The USB was plugged in to a Raspberry Pi 3B+ and booted up performing the initial boot tasks. The boot was not successful, as instead of booting into the GUI, the system was exited out into the terminal (with a few errors i didn't investigate). Upon rebooting the Pi The system was stuck in a Black Screen with no response to user input. Suspecting compatibility issues with raspberry Pi and the Raspbian Buster image, I plugged the USB back into windows to attempt flashing another Raspbian image.

Now up until this point windows is still detecting the USB drive.

I proceeded to open the windows disk management tool to delete the partitions etcher had created. I noticed some odd behavior while doing so, (after deleting one of the partitions on the USB Drive several other partitions, 4 or 5 of them, magically popped up in the disk management tool belonging to the same USB Disk ). I apologize for the vagueness as I can not recall my exact steps but just thought to mention this in order to highlight that the problem might not be directly related to etcher.

I believe it is at this stage that the USB Drive became inaccessible.

Currently a drive belonging to the USB Disk shows up in the Windows file explorer, but can not be accessed.

[image]

The USB disk contains no partitions 

[image]

`diskpart` can not perform operations on the Disk

[image]

The USB Disk is detected in a Linux environment with `dmesg` command

[image]

The USB disk is not listed with the `lsblk` command 

[image]

Cannot be accessed with `parted` command

[image]

In truth at this point I have given up, as any of the suggestions offered in similar issues on the etcher repo do not work, and the USB drive is bricked.

Since the USB thumb drive was purchased today 15 minutes before flashing with etcher, I find it odd for the device to fail so quickly. And I feel if i claim warranty on the USB device it would be a little bit dishonest as it is most likely my actions that led to the device becoming unusable.

I still have hope that it is not a hardware fault and that somehow, by deleting the partitions created by etcher, using the windows disk management tools, has led to the drive becoming unusable.

I would appreciate your thoughts if you feel this might be related to the etcher software and any suggestions and advice if you might think there is still something that may be done to save the thumb drive.

Thanking you for your time and with much appreciation,

- **Etcher version:** 
1.5.59
- **Operating system and architecture:** 
Windows 10 Home 64 bit
- **Image flashed:** 
Raspbian Buster
- **Do you see any meaningful error information in the DevTools?** 
N/A

lurch, 2019-10-20:
> Since the USB thumb drive was purchased today 15 minutes before flashing with etcher, I find it odd for the device to fail so quickly.

Out of curiosity, where did you buy it from? What identification (make, model, brand etc.) does it have on the drive and/or packaging? I only ask because `idVendor=ffff` in the dmesg output looks a little bit "unusual" :wink:

syedsameerali, 2019-10-21:
@lurch 

I began to suspect as much shortly after opening this issue, guess its safe to say the problem was never with the etcher software and the device is counterfeit and has failed. My apologies for jumping to conclusions and not realizing this sooner.

The device was purchased from a local store near my house, there are a lot of counterfeit devices circulating in my country and its getting difficult to spot them, I believe the shop vendors don't know themselves they are counterfeit.

On further inspection I found something is not right about the print quality of the packaging and a flimsiness of the device itself (While unplugging it from the USB port the cap of the thumb drive got pulled out instead, guess they forgot to glue it together properly). The chip inside the plastic enclosure was also suspicious looking.  At least I wont feel bad returning it now.

Thanking you very much for replying and confirming this.

For completeness here are the details and a few pictures of the counterfeit device.

[image]

[image]

[image]

[image]

closed 2019-10-21 by syedsameerali