balena-io/etcher #2949: Disable multiple targets by default

thehalftruth, 2019-10-25:
The "feature" to flash to multiple targets at once should be disabled by default. This "feature" can be very dangerous, it's easily possible to overwrite something unintentionally. Since most users don't need this functionality it should be disabled by default. There should be two options to enable it again in the settings, one for enable it permanently, and one for enable it only once.

To the person that want to say "but you always should have a backup", yes, you should, BUT:
EVEN IF YOU HAVE A BACKUP THIS CREATES UNNECESSARY WORK THAT WAS PREVENTABLE!!!
Some people might have a usb drive that is connected permanently with for example clonezilla to create backups, and of course there is no backup of the usb drive because you can just download it again, but it takes time. Also backup software can fail and than you only have an old backup.

skylerspark, 2019-10-27:
Or, just have a stable enough mental state to not click multiple targets.

lurch, 2019-10-28:
Drive multi-selection was also previously discussed in #2379

closed 2020-02-02 by thehalftruth