balena-io/etcher #2950: "The writer process ended Unexpectedly" Error

skylerspark, 2019-10-27:
- **Etcher version: v1.5.59** 
- **Operating system and architecture: Windows 7 64x Bit** 
- **Image flashed: Not even started** 
- **Do you see any meaningful error information in the DevTools?: No, Only raven.js logging stuff about HTML elements, but no errors or anything that I saw was very very important, but ill attach the javascript log below anyways** 
[-1572138787696.log](https://github.com/balena-io/etcher/files/3775429/-1572138787696.log)

I attached the latest 6GB raspbian buster file .img and linked to my SD Card, and next thing I know, it just "stops" and gives me this error right after I clicked **Flash**

my SD Card is empty, its fat32 and is 16gb suppoted.... this is really fricking anoyying considering the amount of work it was getting all the cords for my raspberry pi, and finding all the SD card bullcrap and adapters.,,

Just more problems to add to the list in my opinion.

I thought this program would be able to "Magically inject and flash my sd card with the image file"
because normally windows will fail and give you the error that the file is too big for the SD card (Its limit is 4gb)

frogmannnn, 2019-11-03:
Error caused by Windows Defender Advanced Threat Protection see #2958

darim92, 2019-11-10:
Same Error on Ubuntu 18.04 when trying to flash sd card

skylerspark, 2019-11-12:
> Error caused by Windows Defender Advanced Threat Protection see #2958

@frogmannnn Its fine, this was a while ago, I actually wound up installing my own version of NOOBS and got raspbian pixel desktop installed in a couple hours. No need for etcher if you have NOOBS

jstyles2000, 2019-11-14:
The threads on this topic sends you in circles.  I was experiencing same error on Windows 7, I realized I wasnt on my pc as an administrator - so there wasnt full access to paths that I wouldnt have expected it needed.  Logged in as admin and it worked perfectly.