balena-io/etcher #2952: Synaptic Package Manager - E: Malformed line 1 in source list /etc/apt/sources.list.d/balena-etcher.list (type)

Bill-Lo3, 2019-10-28:
- **Etcher version:**  Etcher 1.5.59
- **Operating system and architecture:**  Parrot Security OS

Trying to open Synaptic Package Manager.

E: Malformed line 1 in source list /etc/apt/sources.list.d/balena-etcher.list (type)
E: The list of sources could not be read.
Go to the repository dialog to correct the problem.
E: _cache->open() failed, please report.

Bill-Lo3, 2019-10-29:
I ended up just removing it for now but would like to try to install it correct this time. If anyone can, lmk why that happened and how to fix it? Thanks

to uninstall
sudo rm /etc/apt/sources.list.d/balena-etcher.list

sudo apt-update

lurch, 2019-10-29:
> The balena-etcher.list file just contains this
https://github.com/balena-io/etcher.git

That's incorrect. See https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

closed 2023-07-12 by dfunckt