balena-io/etcher #2955: Flashing ANY image suddenly not formatting to full capacity

lwskosa, 2019-10-31:
- **Etcher version:** 1.5.60 (balenaEtcher-Setup-1.5.60.exe)
- **Operating system and architecture:** Windows 10, x64
- **Image flashed:** retropie-4.5.1-rpi2_rpi3.img → https://github.com/RetroPie/RetroPie-Setup/releases/download/4.5.1/retropie-4.5.1-rpi2_rpi3.img.gz
- **Image flashed:** ROTT-TheRA-RPI4-Beta3.img → https://drive.google.com/drive/folders/1H1O0Iedy_QcFoe90cEfxWDsNDk9y6qOK
- **The microSD cards are less than a week old, but problem persist with any brand or aged cards.**
_(referring to microSD as SD going forward)_

I would like to start with this image:
[image]
I am aware that windows by default cannot see the EXT file system, only the FAT (Boot) partition of the card.
My issue is **not** " help, my SD card suddenly recognizes only 250 megabytes!"... but rather: "why is this happening all of a sudden?"

I have not had any issues before. Etcher suddenly decided to format my SD card with only 2.16 gigs on the retropie partition. I have had the full size formatted before. I have tested this with multiple SD card capacities:
2 different 128Gb SD cards: same result for both:
[image]
Flashing the RoTT image:
[image]

But in this case, I have more free space after it is finished (so it is overshooting it by a much larger margin than with RetroPie, which only has ~160MB free space after it is done.
[image]

I have tried...
- Flashing .gz file
- Flashing .img file
- multiple restarts
- different version of etcher (1.5.59)
- completely wiping the SD cards (diskpart, etc)

The result was always the same:
- SD card capacity is limited after flashing (not formatting to full capacity)
- Etcher drive selector recognizes the full capacity:
[image] [image]

Any help/advice would be appreciated, thank you!

lurch, 2019-10-31:
This is the way disk images for Raspberry Pi have **always** worked. The raw disk image contains a fixed MBR, and Etcher doesn't do any modification of the MBR, so whatever the partition sizes are in the disk-image, is what partition-sizes you'll end up with on your SD card.
_However_ some Raspberry Pi distros automatically resize the second partition when they first boot up, to then make use of all available space on the SD card. (Raspbian does this automatically, but I dunno about RetroPie or ROTT as I've never used those. In older versions of Raspbian you had to manually select the option to resize the root filesystem in `raspi-config`.)

lwskosa, 2019-10-31:
> This is the way disk images for Raspberry Pi have always worked. The raw disk image contains a fixed MBR, and Etcher doesn't do any modification of the MBR, so whatever the partition sizes are in the disk-image, is what partition-sizes you'll end up with on your SD card.

As mentioned above, i have had no problems flashing images before this started: My 32GB card would become a 32GB storage, 128GB  would become 128GB once flashed (minus the boot partition, of course)...It **worked** before, not anymore.

If i format the card for NTFS i would then see the full capacity also. Cleaning the card of any partition information makes no difference either once i flash it.

lurch, 2019-10-31:
> 128GB would become 128GB once flashed (minus the boot partition, of course)

No, it only becomes 128GB after the root (second) partition has been resized (which as I said above, _may_ be done automatically by the OS the first time it boots). The _only_ way that it would use all 128GB immediately after flashing and before first booting, would be if the MBR in the raw disk image specified a 128GB partition, and RetroPie etc. would never distribute disk images with a 128GB partition in the MBR, because then it wouldn't work on any SD card smaller than 128GB.
We can see from your screenshots above that RetroPie wouldn't work on an SD card smaller than 3GB, and ROTT wouldn't work on an SD card smaller than 15GB. But (after flashing and booting the OS on an RPi) it should be possible to expand them both to use all available space on your card.

https://www.google.com/search?q=retropie+resize+partition might provide you some useful info?

TotallyNotElite, 2019-10-31:
This is just how raspbian images work. Raspbian automatically resizes the partitions on the first boot. 
[image]

lwskosa, 2019-10-31:
@lurch 
What I meant to say with that one is that just a week ago i did not have to do anything about it, it always allocated the maximum size possible to the retropie partition. I was aware of your suggestion (thank you!) for the partition resizing, but it still is confusing to me as to why it suddenly started allocating thing that way.

lurch, 2019-10-31:
> it still is confusing to me as to why it suddenly started allocating thing that way.

Maybe that's a question for the RetroPie developers? :man_shrugging:

lwskosa, 2019-10-31:
@lurch 

> Maybe that's a question for the RetroPie developers?
That could be, but i am trying to burn the same image i had downloaded previously, it had no issues before.

lwskosa, 2019-11-02:
@lurch 

Update:
i just want to confirm that manually expanding the card to full capacity worked through  raspi-config. (I knew about it, but still made me wonder why wasn't it doing it any more by default as it did before **shrug**)

closed 2019-11-06 by zvin