balena-io/etcher #2956: Broken update version logic

eugenemedvediev, 2019-10-31:
- 1.5.59 
- MacOS Catalina 10.15.1 
- Custom based on 
- Update for version 1.5.59 is not available (latest version: 1.5.60, downgrade is disallowed).
So according to application logic, 1.5.60 is lower than 1.5.59 which is wrong

thundron, 2019-11-04:
That's the expected behaviour: not all versions go through the auto-updater, only those that we tag for release

closed 2019-11-04 by thundron

FlorinAndrei, 2019-11-04:
> only those that we tag for release

@thundron 

Which are those?

I downloaded the app from its main website, installed it, expected it to auto-update - never did. I had to update it manually.

eugenemedvediev, 2019-11-08:
It was an obvious message from balenaEtcher logs that it does not allow DOWNGRADES. Does it mean 1.5.60 is Downgrade for 1.5.59 ???

FlorinAndrei, 2019-11-09:
Guys, all this "some versions are updated, some are not" business is quite confusing to regular users.

Could y'all switch to a simpler upgrade / version policy? Thanks.

lurch, 2019-11-09:
IIRC this policy was chosen so that if there were any accidental bugs in a new release of Etcher, not everyone would be immediately auto-updated to the (possibly buggy) version automatically. I _think_ once a version has proven itself to be "reliable enough", they then push that version into the auto-update process :grey_question: 
But I don't work for Etcher, so I can't be certain.

@thundron I expect people are getting confused by different versions of Etcher being available in different places. Maybe if there was a publicly-visible list (e.g. on the [wiki](https://github.com/balena-io/etcher/wiki) ?) of which Etcher versions are available where, _maybe_ people would be less confused? :man_shrugging:

npmstart-pray, 2019-11-26:
> > only those that we tag for release
> 
> @thundron
> 
> Which are those?
> 
> I downloaded the app from its main website, installed it, expected it to auto-update - never did. I had to update it manually.

same here...three times now.  and I've got all three in my /applications menu.  they lock my machine, forcing me to reboot.

lurch, 2019-12-02:
> they lock my machine, forcing me to reboot.

Maybe you should create a new issue for reporting that?

lurch, 2019-12-06:
I believe #2996 is a good example of why each new version is not pushed through auto-update automatically :wink: