balena-io/etcher #2957: Unexpected errors on MacOS 10.15.1

wpzhhhhh, 2019-11-01:
- **Etcher version:1.5.0** 
- **Operating system and architecture:MacOS 10.15.1** 
- **Image flashed: Raspbian Desktop** 
- **Do you see any meaningful error information in the DevTools?**
[image]

I've tried solution in #2954 and #2833, put iso file into balenaEtcher.app but an unexpected error shows.

simonmeusel, 2019-11-01:
`sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`

lurch, 2019-11-02:
I believe the Etcher developers are already working on fixes for macOS Catalina?
#2939 #2953

flakerimi, 2019-11-05:
This just happened to me.  

sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

Will ask you if you allow Catalina to access USB Drive. After clicking ok will continue.

Look for something like this:

[image]

zvin, 2019-11-06:
Duplicate of #2833

closed 2019-11-06 by zvin