balena-io/etcher #2959: Slax Linux iso inaccurately reported as not bootable

bland328, 2019-11-02:
- Etcher 1.5.59
- Windows 10 Pro x64
- Slax Linux distro (slax-64bit-9.11.0.iso at http://ftp.sh.cvut.cz/slax/Slax-9.x/slax-64bit-9.11.0.iso)
- DevTools:
`Missing partition table ({"image":{"size":277708800,"name":"slax-64bit-9.11.0.iso","path":[REDACTED],"extension":"iso"},"applicationSessionUuid":"41a7af7f-458a-43e3-943f-e3b21e68dabe","flashingWorkflowUuid":"c660c77c-d2cf-4e16-a624-8b7a3deac492","sample":0.1})`

**Problem:**
Etcher warns "It looks like this is not a bootable image," but it is, and the resulting USB flash drive _does_ boot.

bland328, 2019-11-02:
Nothing to see here--the resulting flash drive does _not_ boot. My mistake.

closed 2019-11-02 by bland328

lurch, 2019-11-03:
I guess it must be a "regular" ISO (only boots from CD/DVD) and not a "hybrid" ISO (boots from both CD/DVD and also USB stick) https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos
:slightly_smiling_face:

aethernet, 2023-10-16:
Exactly the reason it's not bootable.

We suggested to Slax maintainer to make it hybrid but he declined.