balena-io/etcher #2960: balenaEtcher is barely accessible on Windows 10 with NVDA

RichMorin, 2019-11-03:
- **Etcher version:** current?
- **Operating system and architecture:** Windows 10
- **Image flashed:** NA
- **Do you see any meaningful error information in the DevTools?** NA

balenaEtcher claims to: "Flash OS images to SD cards & USB drives, safely and easily".

I've used it to write several images to microSD cards and USB flash drives.  It worked very nicely for me, but then I'm a sighted macOS user.  I strongly suspected that it's less convenient and pleasant for blind users, but I don't have any way to assess this. I also don't know how the user experience varies on Linux, macOS, and Windows.

So, before filing this issue, I asked for some help in testing the tool's accessibility.  Specifically, I tried to find some blind folks who would be willing to download, install, and try to use it. Today, I received this informative summary from a friend:

> I downloaded and installed this on Windows 10 with NVDA. The download and install went without a hitch. I launched it from the shortcut that got installed on the desktop. At first it wouldn't talk, but after dismissing a windows warning about letting Java tunnel through the Windows firewall, I was able to navigate the app window with NVDA.

> There are a number of links that are unlabeled, so I don't know what they do. However, I was able to browse my file system to select an image. I don't have an SD card or reader, so I didn't go all the way through the process, but the Flash button is readable and available. Presumably after selecting an image, if I had an SD card mounted I'd be able to press the Flash button and finish the process.

> In other words, it doesn't look completely accessible, but it seems usable in a pinch on Windows 10 with NVDA.

If anyone has follow-up questions, let me know. I'm sure my friend would be willing to provide more information.

-r

lurch, 2019-11-03:
See also #1735 #1734 & #1721

thundron, 2019-12-11:
We're doing a lot of changes (see https://github.com/balena-io/etcher/pull/2987) that will initially limit the experience in this situation, *but* will also let us speed up the process of adding more useful stuff and better reason about a UI workflow "schema" so that we'll be able to add back all the accessibility features, hopefully better than the previous version(s)

RichMorin, 2020-02-27:
Ping.

RichMorin, 2020-03-01:
As a side note, the use of white text on an orange background (in the "Flashing" progress bar) has very poor contrast. It was hard for me to read and would be impossible for users with certain forms of low vision.