balena-io/etcher #2962: gz "Archive Corrupted"

maxgrass, 2019-11-04:
[image]

- **Etcher version: 1.5.60** 
- **Operating system and architecture: macos 10.15** 
- **Image flashed:LibreELEC-RPi2.arm-9.1.502.img** 
- **Do you see any meaningful error information in the DevTools?** 

{"stack":"Error: EPERM: operation not permitted, open '/Users/bla/Downloads/LibreELEC-RPi2.arm-9.1.502.img.gz'","message":"EPERM: operation not permitted, open '/Users/bla/Downloads/LibreELEC-RPi2.arm-9.1.502.img.gz'","name":"Error","code":"EPERM","syscall":"open","errno":-1,"image":"LibreELEC-RPi2.arm-9.1.502.img.gz"}

Mac Security:
Balena has access to removable media, setting full disk access doesn't help.
The problem exists for months now.

lurch, 2019-11-05:
> Operating system and architecture: macos 10.15

Duplicate of #2833 ?

zvin, 2019-11-06:
Duplicate of #2833

closed 2019-11-06 by zvin