balena-io/etcher #2967: After using etcher I now have a signature collision problem with both of my removable USB Drives

orangeme749, 2019-11-12:
- **Etcher version:** newest etcher version
- **Operating system and architecture:** windows 10
- **Image flashed:** tails os 4.0
 
i can roughly remember how the error looked like but dont remember what it said.

lurch, 2019-11-12:
I'm sure the Etcher developers would find a screenshot very helpful, if you're able to reproduce the problem?

orangeme749, 2019-11-12:
[image]
[image]
[image]

orangeme749, 2019-11-13:
I have finally fixed this problem. If you have the same problem as me then follow these steps, they may help.
Open up "CMD" 
- type Diskpart 
- list disk 
- select disk (the offline disk/the has the same gb as your usb device)
 - uniqueid disk 
- (if your disk id looks something like "{58B81DA0-8B1E-4269-9C39-FE5C7B9B58A3}" 
then try - uniqueid disk set=baf784e7-6bbd-4cfb-aaac-e86c96e166ee
 - and your disk id should be changed. 
try to see if it works now? if not then
 try Open a command Prompt as administrator (cmd.exe)
Type Diskpart and press Enter
Type List Disk and press Enter
Type Select Disk X (the offline disk/the has the same gb as your usb device) and press Enter
Type Clean and press Enter
Type Create Partition Primary and press Enter
Type Active and press Enter
Type Format fs=Fat32 Quick and press Enter
Type Exit and press Enter
(THIS WILL DELETE ALL YOUR DATA ON THE USB DEVICE)
I'm not a programmer or a person that works in code or hardware stuff like this but I've just figured it out somehow and wanted to share since I didn't find anything that matched my problem.

closed 2019-11-13 by orangeme749