balena-io/etcher #2969: Built-in information would be welcome

Ricky-Tigg, 2019-11-13:
_Tested:_  balenaEtcher-1.5.63-x64.AppImage under Wayland server.

_Enhancement request_ –  Under icon illustrating  "**?**"are traditionally expected **built-in** information. Access to that information relies on an established internet connection.

[image]

Be the product targeted for enterprise or home uses, at the very least, presence of categories related  to _[licence](https://github.com/balena-io/etcher/blob/master/LICENSE)_, _version_, and _architecture_ would be relevant.  _Version release date_, _OS_ and _Help_, while a bit less relevant, are worth to be mentioned as well.

_Help_ would mention that the software in its free variant, does not support multi-boot to device partition.

lurch, 2019-11-13:
I _think_ you're asking for an offline help page, rather than the :grey_question: icon taking you to an online URL, as it does currently?
Or is this some Wayland-specific request? :man_shrugging: 

> software in its free variant

AFAIK there isn't any other variant??

Ricky-Tigg, 2019-11-14:
Information to be available without **to have** to rely on an established internet connection; that would be what you mentioned as _off-line_. Mention of _Wayland_ was there in case a developer would have needed such information. On that [page](https://www.balena.io/etcher/pro), variant labelled "Pro" was exhibited.

lurch, 2019-11-14:
> On that page, variant labelled "Pro" was exhibited.

AIUI "Etcher Pro" is the hardware that they're building, but it'll still run the same (OpenSource) software as "regular Etcher"? ping @thundron @zvin

Ricky-Tigg, 2019-11-14:
I red there statement "Write to multiple cards or usb disks at once" fast enough to misinterpret it for  "Write to same USB device multiple ISO files at once."; then under _Help_, suggestion "_would mention that the software in its free variant, does not support multi-boot to device partition_" was irrelevant.

lurch, 2019-11-14:
Perhaps this is a duplicate of #2869 ?

Ricky-Tigg, 2019-11-14:
Since my previous comment there was no more object related to _Help_.

lurch, 2019-11-14:
Sorry, I should have clarified - both #2869 and #2969 seem to be related to the functionality of the "?" button? :man_shrugging:

Ricky-Tigg, 2019-11-14:
Don't be sorry, nothing bad! If for maintenance consideration classifying present report as duplicate nay serve your purpose, it has to done then.

closed 2022-03-14 by Ricky-Tigg