balena-io/etcher #2970: Flash instantly fails with linux AppImage and some function in ~/.bashrc

thomas-riccardi, 2019-11-20:
- **Etcher version:** 
./balenaEtcher-1.5.63-x64.AppImage
- **Operating system and architecture:** 
ubuntu 18.04.3 amd64
- **Image flashed:** 
raspberrypi3-2.43.0+rev1-v10.2.2.img.zip

- **Do you see any meaningful error information in the DevTools?** 
```
error 
OperationalError {cause: Error: Command failed: cd "/home/thomas/Downloads"; "/usr/bin/pkexec" --disable-internal-agent /bin/…, isOperational: true, killed: false, code: 2, signal: null, …}
cause
:
Error: Command failed: cd "/home/thomas/Downloads"; "/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT; sh '/tmp/tmp-19334YmI5Y68PjJ8v.cmd'" /tmp/tmp-19334YmI5Y68PjJ8v.cmd: 113: export: BASH_FUNC_grep_custom%%: bad variable name at ChildProcess.exithandler (child_process.js:303:12) at ChildProcess.emit (events.js:182:13) at maybeClose (internal/child_process.js:961:16) at Socket.stream.socket.on (internal/child_process.js:380:11) at Socket.emit (events.js:182:13) at Pipe._handle.close [as _onclose] (net.js:596:12)
cmd
:
"cd "/home/thomas/Downloads"; "/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT; sh '/tmp/tmp-19334YmI5Y68PjJ8v.cmd'""
code
:
2
isOperational
:
true
killed
:
false
signal
:
null
message
:
"Command failed: cd "/home/thomas/Downloads"; "/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT; sh '/tmp/tmp-19334YmI5Y68PjJ8v.cmd'"↵/tmp/tmp-19334YmI5Y68PjJ8v.cmd: 113: export: BASH_FUNC_grep_custom%%: bad variable name↵"
name
:
"Error"
```

The interesting part is:
```
/tmp/tmp-19334YmI5Y68PjJ8v.cmd: 113: export: BASH_FUNC_grep_custom%%: bad variable name
```
The generated tmp cmd file contains a list of variables `EXPORT` coming from my `~/.bashrc`:
```
export BASH_FUNC_declare_grep_exclude_arguments%%='() { <some work> }'
```

In my `~/.bashrc`:
```
function grep_custom() { <some work> }
export -f grep_custom
```

The `export -f` seems to break the script generation which assumes `sh` and not `bash`.

thomas-riccardi, 2019-11-20:
The .deb from the apt repository documented in the README in github does work. (Alas it was *not* documented in the etcher website, I just found-out it existed.)

lurch, 2019-11-20:
ping @jorangreef as it looks like it might be an issue with sudo-prompt ?  :man_shrugging:

thomas-riccardi, 2019-11-20:
It seems it was in the generated script in `/tmp`, it may have been generated before the sudo prompt...

lurch, 2019-11-20:
@thomas-riccardi I was wondering if your problem is due to https://github.com/jorangreef/sudo-prompt/blob/master/index.js#L128

zvin, 2019-11-21:
@lurch @jorangreef this is not related to sudo-prompt
the script comes from https://github.com/balena-io/etcher/blob/master/lib/shared/permissions.js#L125

thomas-riccardi, 2019-11-22:
Maybe skip `BASH_FUNC_*%%` or `*%%` env vars?

Or maybe just use bash in the generate tmp cmd script? (the sudo-prompt already assumes bash is available). 
But maybe there is a similar issue with other shells where using bash would still not be enough?

zvin, 2019-11-27:
@thomas-riccardi fixed in `v1.5.64`

closed 2019-11-27 by zvin