balena-io/etcher #2974: 110MB .exe that is too bad

slrslr, 2019-11-22:
110MB .exe file of Windows portable Balena Etcher is something i can not comprehend.
Why do you waste data transfer and disk space of all the people that need to use this software.
I was thinking the "dd" tool used on Linux is maybe 1MB. I need to just copy Raspbian .img on SD card so it is bootable. Raspberry site and various tutorials recommend this SW for WIndows and i am quite touched that they recommed bloated software like this. Why i need to download stupid 110MB .exe for the mentioned purpose? Why it is so extremely large "just" to copy .img to a SD card?

lurch, 2019-11-22:
See e.g. #2857

gitmayer, 2019-11-25:
Use win32diskimager. I can't stay on Etcher due to the size of the app.  I can't comprehend where this size comes from except using bloated libraries and not knowing how to compile for a smaller size.  It's a shame because it has a nice interface, but doesn't really do all that much different than other disk imaging programs.

0kulist, 2020-01-28:
~100MB for Mac version, that unpacks to 260MB. Are you ... serious ? It doesn't deserve to be run! Deleted!

ykla, 2020-01-29:
So why it‘s’ so big？

gitmayer, 2020-01-29:
Looks like they're saying it's the framework they are using (from what they've said, they are using the Electron framework) and that they need to trim dependencies.

rradar, 2020-03-19:
@ykla 

> So why it‘s’ so big?

Etcher does a little bit more than *just* writing and verifying your image. It comes included with a lot's of stuff you don't see but which is good for you like tracking and analytics. The ads you see maybe but most of the other ware is under the hood.

It's actually even worse. Beside the 100MB there is another approximately 300MB of dependencies necessary. Other applications like usbimager manage to do the same (writing and verifying) and do not even need 1MB of disk space and no dependencies.

closed 2021-07-13 by thundron

einsteinx2, 2021-11-24:
Just updated from 1.5.x to 1.7.0 and on macOS the app size is now 490MB!! Absolutely insane for such a simple program. Discord, which is also an Electron app but is orders of magnitude more complicated is less than half that size... And the 1.5.x version I updated from was already a ridiculous 223MB. Absolutely incredible...

wwderw, 2021-11-24:
Going to something like Webview would actually help not only the size, but also resource consumption as well.  And still able to use frontend for the UI (which I like frontend and immediate mode over the traditional retained mode).

I do like the app and still use it (although not on the latest and greatest), but I do think that going to Webview may help out greatly with resource usage all the way around.

KubaO, 2025-03-19:
To put into perspective how ridiculously big it is:

It is bigger than the complete Python 3.13.2 installer. Balena itself is the equivalent of a 2500-line Python program. So you could literally have a cross-platform solution in about 30MB for Windows, 70MB for MacOS, and you'd get an entire Python install for free.

Of course, nobody needs to distribute the whole Python with such a simple app. `pyinstaller` would make it 10MB or so. And even that is ridiculously large.

pschonmann, 2025-03-20:
You're welcome to use alternatives
https://alternativeto.net/software/etcher/