balena-io/etcher #2975: Weird password prompt on macOS Catalina

jwa5426, 2019-11-24:
- **Etcher version:** 
1.5.63
- **Operating system and architecture:** 
macOS Catalina 10.15.1 (19B88)

Is this password prompt normal? It doesn't look like any prompt I've seen from MacOS.

Typical password prompt:
[image]

Etcher password prompt:
[image]

What's the deal with this? Doesn't seem like a major issue but it looks weird enough to be a phishing attempt or something.

sneak, 2019-11-26:
Good catch.  I wondered about this too, but I have all network disabled for etcher because of #2977 so I didn't follow up.  Subscribing to see what comes of this.

sneak, 2019-11-26:
https://github.com/balena-io/etcher/blob/75e6f1e39a484838aad99f903c0ade86f027d3d9/lib/shared/catalina-sudo/sudo-askpass.osascript.js#L10

Looks like they're firing off a local application dialog (not a privileged dialog) as a separate process via `sudo` by setting `SUDO_ASKPASS`:

https://github.com/balena-io/etcher/blob/75e6f1e39a484838aad99f903c0ade86f027d3d9/lib/shared/catalina-sudo/sudo.js#L22

This is definitely sketchy, and it should not be done this way, not the least of which is because it's a very generic "wants to make changes", but also for the reasons cited above.  This trains the (potentially novice) user to type their admin password into a sketchy nonstandard dialog—terrible security practice.

thundron, 2019-11-28:
It's because of Catalina's new permission system, unfortunately there is no better option as of now but we're constantly checking for a better solution.

closed 2019-11-28 by thundron

ajnsit, 2019-12-26:
This is definitely sketchy. Is it possible to leave this issue open until a better solution is found?