balena-io/etcher #2976: The writer process ended unexpectedly.

koumakpet, 2019-11-25:
- **Etcher version: 1.5.45** 
- **Operating system and architecture:**  Windows 10 Home 64-bit
- **Image flashed:**  kali-linux-2019.3a-rpi3-nexmon-64.img
- **Do you see any meaningful error information in the DevTools?**  No, not really

I have tried to flash multiple images, but it always just gave me an Error after the administrator prompt

Full Error message:
> The writer process ended unexpectedly. Please try again, and contact the Etcher team if the problem persists.

vinyll, 2019-11-27:
Updating solved for me.
If not, run the app with `sudo`

thundron, 2019-11-28:
Hi, could you please open the devtools console when that happens?
It contains the logs of what's happening inside the application, so it's useful if you could share those

Pritel03, 2019-12-14:
[-1576346119910.log](https://github.com/balena-io/etcher/files/3964135/-1576346119910.log)

nerdybeast, 2020-01-13:
I'm on Windows running Etcher with admin rights and I too am getting the `Error: EPERM: operation not permitted, write","code":"EPERM","syscall":"write"` error

kunalspathak, 2020-02-15:
Seeing the same error. I tried adding balena-etcher folder as exclusion from Windows Defender antivirus scans and turning off "Virus and threat protection". However I continue get this error. Any other way to diagnose what is going on?

lurch, 2020-02-16:
@kunalspathak See @thundron 's comment above :slightly_smiling_face:  (@thundron is one of the Etcher developers)

billagee4, 2020-04-17:
I have the same problem, trying to flash a 64gb SD card with the 64bit Kali Linux image for Raspberry Pi 4.  It starts and some stuff is copied to the SD card, then it prematurely ends and gives the error mentioned.  I'm using OSX Catalina.  Is there a fix?  Thanks

From the log:
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:27705 Download the React DevTools for a better development experience: https://fb.me/react-devtoolsYou might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.76, Type = dmg
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Fri Apr 17 2020 09:14:00 GMT+0800 (Hong Kong Standard Time) Application start ({"packageType":"dmg","version":"1.5.76","applicationSessionUuid":"8f9f70e3-ade8-40bc-988e-4692436ecc00","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Fri Apr 17 2020 09:14:29 GMT+0800 (Hong Kong Standard Time) Select image ({"image":{"size":6999999488,"name":"kali-linux-2020.1a-rpi3-nexmon-64.img","hasMBR":true,"partitions":[{"offset":512,"size":128000000,"type":12,"index":1},{"offset":128000512,"size":6870999552,"type":131,"index":2}],"path":"/Users/Family/Downloads/kali-linux-2020-1a-rpi3-nexmon-64-img-xz/kali-linux-2020.1a-rpi3-nexmon-64.img","extension":"img","logo":false,"blockMap":false},"applicationSessionUuid":"8f9f70e3-ade8-40bc-988e-4692436ecc00","flashingWorkflowUuid":"ff19e033-e0ae-4795-b9cf-86d3a823d3de","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Fri Apr 17 2020 09:14:36 GMT+0800 (Hong Kong Standard Time) Flashing 0 devices, 0% at null MB/s (total null MB/s) eta in undefineds with 0 failed devices
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Fri Apr 17 2020 09:14:36 GMT+0800 (Hong Kong Standard Time) Flash ({"image":"/Users/Family/Downloads/kali-linux-2020-1a-rpi3-nexmon-64-img-xz/kali-linux-2020.1a-rpi3-nexmon-64.img","drives":[{"size":62587404288,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"Secure Digital","isCard":true,"isUSB":false,"devicePath":"IODeviceTree:/PCI0@0/RP04@1C,3/SDXC@0,1","mountpoints":[{"path":"/Volumes/NO NAME","label":"NO NAME"}],"description":"Apple SDXC Reader Media","isUAS":null}],"driveCount":1,"uuid":"17595c6e-249c-48b7-901b-5c2437c2386f","status":"started","flashInstanceUuid":"17595c6e-249c-48b7-901b-5c2437c2386f","unmountOnSuccess":true,"validateWriteOnSuccess":true,"trim":false,"applicationSessionUuid":"8f9f70e3-ade8-40bc-988e-4692436ecc00","flashingWorkflowUuid":"ff19e033-e0ae-4795-b9cf-86d3a823d3de","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Elevating command: /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper (Renderer).app/Contents/MacOS/balenaEtcher Helper (Renderer) /Applications/balenaEtcher.app/Contents/Resources/app
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Fri Apr 17 2020 09:14:37 GMT+0800 (Hong Kong Standard Time) SafeWebview loaded ({"event":{"id":1,"url":"https://assets.balena.io/etcher-featured/index.html?etcher-version=1.5.76&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1587086077092.088,"resourceType":"mainFrame","ip":"104.31.83.157","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["560"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["58524a4daa9fd1e3-HKG"],"cf-request-id":["02274cc48b0000d1e335a63200000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Fri, 17 Apr 2020 01:14:37 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Fri, 17 Apr 2020 00:54:10 GMT"],"last-modified":["Thu, 16 Apr 2020 21:26:05 GMT"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["1"],"x-fastly-request-id":["0de64af6ecc233ffda14bdc7fa6844f4e26b01ed"],"x-github-request-id":["4566:6F5B:7FFC1:A1FD6:5E98FBD9"],"x-proxy-cache":["MISS"],"x-served-by":["cache-hkg17928-HKG"],"x-timer":["S1587086077.084586,VS0,VE1"]},"referrer":"","error":"net::OK"},"applicationSessionUuid":"8f9f70e3-ade8-40bc-988e-4692436ecc00","flashingWorkflowUuid":"ff19e033-e0ae-4795-b9cf-86d3a823d3de","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 Successfully connected to IPC server: etcher-server-2861, socket root /var/folders/js/11r6_5xj6_50nsvz4mw077rc0000gn/T/
[... cut ...] ded unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"code":"ECHILDDIED"}
console.<computed> @ /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58
logException @ /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/resin-corvus/src/resin-corvus.js:233
(anonymous) @ app.ts:45
wrapped @ /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:444

NikolaMaroski, 2020-05-11:
I resolved this issue by allowing the balenaEtcher app (instead Rufus) through Controlled folder access
Windows Security -> Controlled folder access -> Manage Controlled folder access -> Allow an app through Controlled folder access

koumakpet, 2020-05-13:
I don't use windows anymore and so the issue is not relevant to me, It seems like @NikolaMaroski managed to find the answer so I'm closing it, but thanks for the help everyone.

closed 2020-05-13 by koumakpet

theronic, 2020-07-07:
This hit me today. I had to downgrade from v.1.5.x to v1.4.9 to get it working on OSX Mojace 10.14.5, but I noticed that a bzipped .bz2 image file (.img) worked, so I suspect a bug in the handling code for non-compressed images.

lurch, 2020-07-07:
> I suspect a bug in the handling code for non-compressed images.

That seems unlikely - I suspect the handling code for non-compressed images is actually _much_ simpler than the handling code for compressed images. But I'm not an Etcher developer :man_shrugging:

thundron, 2020-07-08:
@theronic which version of 1.5.x gave you issues? Can you try with the latest?