balena-io/etcher #2977: Etcher spies on the user without consent.

sneak, 2019-11-26:
Reopening #2051 - the issue is still valid.

- **Etcher version:** 1.5.63
- **Operating system and architecture:** darwin x64

Wikipedia defines spyware as:

> Spyware is a software that aims to gather information about a person or organization, sometimes without their knowledge, and send such information to another entity without the consumer's consent. 

Etcher has done *precisely this* for some time.

This silent tracking includes IP address and timestamp information, which is more than sufficient to identify a user (and perhaps even their physical location) to the other people who gain access to this data, such as analytics providers, network hosts, interchange points, ISPs, and intelligence services.

Upon opening Etcher for the **first time**, the following connections are attempted:

[image]
[image]

At no point am I prompted for consent, or provided the ability or UI to opt out.  This happens *silently*, regardless of user intent or consent.  Only after balena has been contacted by the software does the main application window open:

[image]

Then, the user could attempt to disable the settings, but by then their IP address (and physical location) has by this time already been transmitted to the manufacturer, likely against their consent and wishes.

This issue is not about the GDPR, or the legality of this collection, simply the very practical issue that the software phones home and **leaks the user's IP address to the developers without consent**.  At no point does the user have the ability to disable this on first launch.

At present, that makes this application fit the definition of **spyware**.

Remember: humans have an inalienable right to privacy.  By leaking users' personal data (even if you do not save it, or don't receive it yourself - by *causing it to be sent out of their computer at all*) you have **infringed upon their human rights**.

Do not abuse the human rights of your users.  Ask the user for consent before transmitting *any* data out of their computer.

m00dy, 2019-11-27:
hmm greek style spyware xD

ukd1, 2019-11-27:
Looks like there were some updates shipped to try stop this a while back;

- https://github.com/balena-io/etcher/blob/a155811678bfaac5439a301bf8e74d1d8b86b0b1/CHANGELOG.md#v121---2017-12-06

but that it's probably this request - https://github.com/balena-io/etcher/blob/a155811678bfaac5439a301bf8e74d1d8b86b0b1/lib/gui/app/modules/analytics.js#L30 -that's causing your alert. 

That url returns;

```json
{
  "version": 1,
  "analytics": {
    "mixpanel": {
      "HTTP_PROTOCOL": "https",
      "api_host": "api.balena-cloud.com/mixpanel",
      "probability": 0.1
    }
  },
  "autoUpdates": {
    "checkForUpdatesTimer": 300000,
    "autoUpdaterConfig": {
      "autoDownload": false
    }
  }
}
```

This should probably not be requested, as it doesn't look like it's needed?

sneak, 2019-11-27:
The line you quoted does not make a network request, it just sets a constant.

cixtor, 2019-11-27:
@sneak you clearly haven’t installed + used `brew cask install steam`

I have thirty-eight (38) hostnames blocked for “Steam.app” and twelve (12) for “Steam Helper.app”.

Blocking one or two requests from “Balena Etcher” is not a big deal compared to other apps.

sneak, 2019-11-28:
Comparing an unethically-produced, dangerous app to another unethically-produced, dangerous app does not make either one good or safe.

cixtor, 2019-11-28:
@sneak how is Steam an unethically-produced / dangerous app?

cixtor, 2019-11-28:
> _The issue is not for people like you who block it, obviously - you are not affected by it. You are an outlier. The issue is for the thousands and thousands of people who are silently being spied on without their knowledge and without their consent._

Fair enough. I understand your point of view now.

thundron, 2019-11-28:
As you may or may not know, all data Etcher gathers is anonymous which doesn't violate any "user" or "person" data (as in, there's nothing in our analytics that points to you specifically)
We had a discussion for opt-in vs. opt-out a while ago, but we still don't see enough reasons for us to switch over to opt-in, even more so since if it's really an issue, you can disable the analytics in the settings and the only calls you'll see will be for external content that is better placed outside of the application (i.e. the featured project)

closed 2019-11-28 by thundron

sneak, 2019-11-29:
> * As you may or may not know, all data Etcher gathers is anonymous which doesn't violate any "user" or "person" data (as in, there's nothing in our analytics that points to you specifically)*

This is a false statement. The data Etcher transmits from my machine includes my IP address, which uniquely identifies me.

evanlandreneau, 2019-11-29:
Looks like it's time to move to UNetBootin...

lurch, 2019-12-02:
@sneak Out of curiosity, how did you download Etcher in the first place without "leaking your IP address and physical location to thousands of people" ??

sneak, 2019-12-04:
I don’t use Etcher (funny because it seems I will likely soon be forking it); I know how to use dd. Anyone who wishes, however, can download Etcher using Tor and it will not disclose their ip/location/identity to the manufacturer.

jordanius, 2020-03-07:
You could always just run Tor on your router, and then ditto for any phone-home telemetry by any program. Problem solved.

Incidentally, I wonder why you're trying to pick on this one application when, to the best of my knowledge, at least one out of four Mac apps "phones home" with similar telemetry. Of course you're going to say that just because others are doing it too doesn't make it ok, but in any case, it seems you really have your work cut out for you, and had better get to writing all of the services promoting all of those other apps too. Or, actually, on second  thought, please don't. Most people aren't really bothered by this sort of telemetry. For those who are, like you, there's always the route of installing Tor directly on your router and having all outgoing connections that you make from any app anonymized. So just solve your own problem, and quit bothering other people who couldn't care less about this "invasion of their privacy" (or if they did care, they probably already thought of the Tor-router thing and solved their own problem themselves, unlike you).

pinchies, 2020-03-07:
"most people aren't really bothered" -> Citation needed. AFAIK, most people aren't aware. If that were true, and most people really weren't bothered, then why not _ask_ before transmitting?
GDPR set the standard to be expected.

sneak, 2020-03-07:
It is unethical to assume consent.

rradar, 2020-03-07:
Easy nowadays. Don't use software which comes bundled with adware and spyware (namely balena etcher).

You can have a 'clean' version of etcher which is called [**usbimager**](https://gitlab.com/bztsrc/usbimager), does things right, respects your privacy and comes in a executable package of 250kb :tada:

sneak, 2020-03-07:
usbimager's UI is super confusing, especially for new users who just want the 1-2-3 of "select image, select disk, hit flash".  There's a reason people recommend Etcher.  The correct answer is to fork this free software, rename it, remove the spyware, and re-release it.

shideneyu, 2020-03-31:
Popups started to come up BECAUSE of this dumpy software. That's why I hate Windows. Thank you, Belena, for installing all of those adware on my friend's machine.

thundron, 2020-04-02:
@shideneyu Etcher doesn't install any adware and will never do. Either your friend installed some bloatware coming from another website/application, downloaded from a *highly* untrusted source or has something else on his PC that is causing trouble.
Either way, Etcher isn't the issue and is definitely not spying on anyone's PC.
There is an option in the settings that disables the *anonymous* data collection (which means there's nothing related to you but only how the application is working, errors and the likes) - as such, there's no need to continue the conversation any further.