balena-io/etcher #2978: Can't install on Raspberry Pi 3B+

hageson, 2019-11-28:
- Etcher 1.5.64-ia32?
- Raspbian Buster September 2019 

I tried installing on my Raspberry 3B+following the instructions for Debian and Ubuntu distributions.  It failed.  Here is the terminal dialog:

`pi@rp1:~ $ echo "deb https://deb.etcher.io stable etcher" | sudo tee /etc/apt/sources.list.d/balena-etcher.list
deb https://deb.etcher.io stable etcher
pi@rp1:~ $ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.30RnouXJjy/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: key 379CE192D401AB61: public key "Bintray (by JFrog) <bintray@bintray.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
pi@rp1:~ $ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                
Ign:3 https://dl.bintray.com/etcher/debian stable InRelease                    
Get:4 https://dl.bintray.com/etcher/debian stable Release [3,674 B]
Get:5 https://dl.bintray.com/etcher/debian stable Release.gpg [821 B]
Fetched 4,495 B in 2s (2,363 B/s)  
Reading package lists... Done
**N: Skipping acquire of configured file 'etcher/binary-armhf/Packages' as repository 'https://deb.etcher.io stable InRelease' doesn't support architecture 'armhf'**
pi@rp1:~ $ sudo apt-get install balena-etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package balena-etcher-electron
pi@rp1:~ $ 
`

hageson, 2019-11-29:
I realize now that I don't need Etcher on the Raspberry - I can accomplish what I need using the built-in SD copy utility.

closed 2019-11-29 by hageson

lurch, 2019-12-02:
Duplicate of #2870