balena-io/etcher #2994: Enable unsafe mode window off screen

jpmcb, 2019-12-04:
- **Etcher version:** 1.5.66
- **Operating system and architecture:** MacOS - 10.14.6 x64
- **Image flashed:** N/a
- **Do you see any meaningful error information in the DevTools?** No console error or any indication of an error

**To reproduce:** Settings -> Unsafe mode button

Window appears off screen. Screenshot related.

[image]

thundron, 2019-12-09:
Can you try with the latest v1.5.67 @jpmcb ?

jpmcb, 2019-12-09:
I've updated to v1.5.68 and it appears that this is _not_ fixed:

[image]

Please let me know if there's anything else I can do to help you hunt this bug down!

thundron, 2019-12-11:
Sure (thank you a lot!), next step would then be to open the devtools console and check the logs after opening the confirmation dialogue, to check if there's any error/useful logs (devtools console on macOS can be opened with `cmd + opt + i`) and paste them here

jpmcb, 2019-12-11:
```
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/resin-corvus/src/resin-corvus.js:39 
Wed Dec 11 2019 09:27:27 GMT-0700 (Mountain Standard Time) Toggle setting ({"setting":"unsafeMode","value":false,"dangerous":true,"applicationSessionUuid":"83e056fd-97d0-40f8-b725-7230ba639f83","sample":0.1})
```
This is all that appears in the console

thundron, 2019-12-11:
Found the issue, it's gonna be fixed with #2987

closed 2019-12-13 by thundron