balena-io/etcher #2996: Regression: 1.5.66: BeagleBoard.org images (ext4 *.img, compressed with xz), 1.5.65 Works fine..

RobertCNelson, 2019-12-04:
- 1.5.66
- Debian Buster (10.x)
- https://rcn-ee.net/rootfs/bb.org/testing/2019-12-01/stretch-iot/bone-debian-9.11-iot-armhf-2019-12-01-4gb.img.xz
- nope: (attached)

[image]

gcgarner, 2019-12-06:
Hello

I have the same issue trying to write a Rasbian image to an sd card. I get this if i try to burn the zip or the img file. I verified that the SHA-256 of the download and it is correct.

Version 1.5.66
 
```bash
$ uname -a
Linux pop-os 5.3.0-22-generic #24+system76~1573659475~19.10~26b2022-Ubuntu SMP Wed Nov 13 20:0 x86_64 x86_64 x86_64 GNU/Linux
```
Please let me know if you need any log files to assist with the issue.

roman-mazur, 2019-12-10:
We reproduced a problem on Linux in the new Etcher version which leads to this error message in the UI.

More details on the error can be found in dev console log:
```
{“stack”:“Error: The elevated process died unexpectedly\n at Object.exports.createError (file:///tmp/.mount_balenaf8fehu/resources/app.asar/generated/gui.js:1797:17)\n at Object.exports.createUserError (file:///tmp/.mount_balenaf8fehu/resources/app.asar/generated/gui.js:1836:18)\n at file:///tmp/.mount_balenaf8fehu/resources/app.asar/generated/gui.js:11381:22”,“message”:“The elevated process died unexpectedly”,“description”:“The process error code was 9”,“report”:false,“image”:“2019-09-26-octopi-buster-lite-0.17.0.img”}
```

The issue should be fixed with #3007

commit c251bce4 (2019-12-10): Merge pull request #3007 from balena-io/fix-appimage

Don't add --no-sandbox when ELECTRON_RUN_AS_NODE is 1
Fixes #2996
closed 2019-12-10 by zvin