balena-io/etcher #2997: Unable to enter administrator password on the new popup prompt after selecting flash “balenaEtcher wants to make changes.”

ldon13, 2019-12-05:
- **Etcher version: 1.5.66** 
- **Operating system and architecture: MacOS Catalina** 
- **Image flashed: Hypriot Docker** 
- **Do you see any meaningful error information in the DevTools? Console tab. No**

thundron, 2019-12-09:
Hi, can you please describe a bit more the issue? Do you mean that you can't actually write the password (as in, you write but nothing happens) or that you can't confirm after writing the password, or something else?

fromanifromage, 2019-12-11:
I might have the same error. First I see this message:
"balenaEtcher wants to make changes. Type your password to allow this."
Then I try to enter the correct password. I have tried alternating both the user and admin passwords; neither works. The program claims it will show a progress bar and it never shows.

ldon13, 2019-12-11:
@thundron, After typing the password nothing happens. I don’t know what the reasoning behind switching to a custom prompt only asking for the password while the default MacOS prompt asks both for username and password. It hasn’t been a problem with the default in the previous versions.

thundron, 2019-12-11:
@ldon13 Reason is macOS Catalina unfortunately 🤦‍♂ 
If you run with `sudo` does it work?
Can you paste here the output in the devtools console *after* closing the password dialogue? (`cmd + opt + i` to open it)

JokerQyou, 2019-12-15:
When running in macOS Catalina 10.15.1, Etcher used to work properly, and a dialog like this would popup before actually flashing to SD card:
[image]

After upgrading to Catalina 10.15.2, Etcher changed to this popup:
[image]

And whatever you type in this dialog does not work, even if the password is correct. If you click "OK", it will popup again; if you click "Cancel", well, it cancels the flash.

lurch, 2019-12-16:
> I did not find related description in the changelog

I think #2939 is the relevant PR ? Which only appears in the changelog as https://github.com/balena-io/etcher/blob/master/CHANGELOG.md#v1561

lurch, 2020-09-17:
AFAICT, this sounds like the problem I described here: https://github.com/balena-io/etcher/issues/3107#issuecomment-602047807

JokerQyou, 2020-09-17:
> This comment says that even PrivilegedHelperTools (whatever they are) don't work in Catalina either.

I guess not. ApplePiBaker uses privileged helper tool, but it's working all the time. I installed it last year (you can see my comment above) and the last time I used it two weeks ago everything works fine. I've long been using Catalina (currently on 10.15.6)

shawaj, 2020-12-13:
This should be fixed now in v1.5.112. @Decrypt0r @JokerQyou @fromanifromage could you update to the latest version and check if it now works for you?

sjohnson-imi, 2021-01-26:
The issue is still occurring. I'm using v1.5.115 on Catalina. It only asks for a password (not an administrator login) and then silently kicks back to the "select target" decision. I tried repeating this thrice (and on to five times) to see if that would get by, but no change.

sjohnson-imi, 2021-01-26:
Seems to stem from the change from asking for admin credentials to asking for user credentials and relying on sudoers to allow access.  Once I added the standard user account to the sudoers file, etcher was able to complete.

MennoLabs, 2021-08-30:
I am using 1.5.121 on Kubuntu 21.04.
After selecting a target, a password prompt pops up and disappears before I can enter a password.
The application then stalls at "Starting...", without ever actually starting.

DavidAmeba123, 2022-03-14:
Is there a default username and password for Etcher? I'm trying to download the setup but they are asking for a username and password before they allow me to download.

mcraa, 2022-03-14:
@DavidAmeba123 the real etcher website is not asking you any password
Please have a look at #3724 and double-check if you are really on https://etcher.io.
When you start a flash with etcher it will ask you for an admin password to have all the rights for writing the target drive.

lurch, 2022-03-14:
> double-check if you are really on https://etcher.io/

That auto-redirects to https://www.balena.io/etcher/ :wink:

jdeibele, 2022-07-22:
I'm on Ventura Public Beta and having a similar problem to what [JokerQyou](https://github.com/JokerQyou) reported back in 2019.  It's worse in that I can't flash.

I run as a user without administrative privileges.   There is a different account with administrative privileges.   Normally, I need to type in the name and password of an account with administrative privileges.

When I try to flash an ISO, I get this:

[image]

Note that it is not asking for the name of an administrative account.  On my computer, there is one account with administrative privileges but there could be 10.

After entering the password for the administrative account 3 times, I get this:
[image]

The "Flash!" button flickers but doesn't do anything.   I choose "Select Target" and repeat the cycle.

kmbgit, 2023-06-15:
This works from a terminal (where "admin" is the name of a user with Administrator permission)...

su admin -c "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher"

Alternatively, download the Raspberry Pi Imager from https://www.raspberrypi.com/software/ - it manages to ask for an admin username/password so it must be possible.

dfunckt, 2023-07-12:
Closing in favor of #3321

closed 2023-07-12 by dfunckt