balena-io/etcher #3006: Etcher loads remote code after privileges are elevated

bboc, 2019-12-08:
- **Etcher version:** 1.5.59
- **Operating system and architecture:** MacOS 10.13.6
- **Image flashed:** Raspbian
- **Do you see any meaningful error information in the DevTools?** n/a

When I flash the image, Etcher prompts me for my password, because needs elevated privileges to for some operations on the SDcard. Right after entering the password, Etcher opens a connection to  jquery.org - probably do download jQuery. This is a major security risk, as it allows for potentially unsafe code to be executed on my system AFTER I elevated privileges.

Note: I have disabled error reporting and usage statistics, so no network connection should be open when I use the program, except for downloading images. I am pretty sure jQuery.org tracks downloads, which may or may not count as "usage statistics" depending on one's font of view.

lurch, 2019-12-08:
AIUI that should be fixed in newer Etcher versions? See https://github.com/balena-io/etcher/issues/2766#issuecomment-531437542 for an explanation.

bboc, 2019-12-08:
Ok. I hope I have a spare SDcard somewhere, so I can test that with the latest version.

bboc, 2019-12-08:
I just confirmed that same behavior is still present in version 1.5.67, which is the version currently available for download.

jQuery appears to be related to the ad on the left side of the windows that is displayed while flashing is in progress. When I block jquery.org, the flashing is unaffected, but the ad is not displayed.

lurch, 2019-12-09:
ping @petrosagg - Has something slipped through the net? Does there need to be a process in place to prevent that happening? :man_shrugging:

thundron, 2019-12-09:
@lurch the EFP does load GA, but only if analytics are enabled in the settings:
```javascript
if (optOutAnalytics !== 'true') {
  // ...
```
Note: disabling analytics doesn't mean disabling network connections, it just means ... disabling analytics, I struggle to understand how the two should be related. Closing for now

thundron, 2019-12-09:
Also just to be super clear - what was unintentional was loading GA regardless/in the success-banner, the EFP actually makes use of that (when it's enabled, mind!)

closed 2019-12-09 by thundron

bboc, 2019-12-09:
@thundron 

> Note: disabling analytics doesn't mean disabling network connections, it just means ... disabling analytics, I struggle to understand how the two should be related.

Maybe I can explain that: Since every network connection inevitably exposes the IP address, the remote end can track the IP and run analytics on that. You have no idea what jquery.org does with the IP. GDPR considers the IP address personal data, so technically you might have to have a data processing agreement with them. 

And besides, even if it's common practice to load 3rd-party code, even on a website this is an unnecessary security risk, there's several ways malicious code can be introduced without anyone noticing. 

> Closing for now

Fine for me.

rradar, 2019-12-15:
@thundron wrote: 
> Note: disabling analytics doesn't mean disabling network connections, it just means ... disabling analytics, I struggle to understand how the two should be related.

@bboc wrote:
> Since every network connection inevitably exposes the IP address, the remote end can track the IP and run analytics on that.

@bboc wrote:
> GDPR considers the IP address personal data, so technically you might have to have a data processing agreement with them.

And **ask for users permission** (alias opt in)