balena-io/etcher #3010: Failed device when writing multiple SD cards at the same time

pglu, 2019-12-11:
- **Etcher version:** 1.5.52
- **Operating system and architecture:** Windows 10
- **Image flashed:** Raw image (.img)
- **Do you see any meaningful error information in the DevTools?**

I work with an ICY BOX with 10 slots for Micro SD cards. In Etcher it is possible to specify several Micro SD cards as target. When writing on the card the error message "... Failed device" appears.

If the parallel writing does not work, it would also be good if it could be done serially automated (i.e. one card after the other).

thundron, 2019-12-11:
Hi, can you fill in the last question of the template regarding devtools console? (`ctrl + shift + i` to open it)

pglu, 2019-12-11:
Please see the last line

Download the React DevTools for a better development experience: https://fb.me/react-devtools
You might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.52, Type = nsis
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:32 GMT+0100 (Mitteleuropäische Zeit) Application start ({"packageType":"nsis","version":"1.5.52","applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:34 GMT+0100 (Mitteleuropäische Zeit) SafeWebview loaded ({"event":{"fromCache":false,"headers":{"Accept":"text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8","Upgrade-Insecure-Requests":"1","User-Agent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.52 Chrome/66.0.3359.181 Electron/3.1.9 Safari/537.36"},"id":11,"method":"GET","referrer":"","resourceType":"mainFrame","responseHeaders":{"Accept-Ranges":["bytes"],"Cache-Control":["public, max-age=0"],"Connection":["keep-alive"],"Content-Encoding":["gzip"],"Content-Type":["text/html; charset=UTF-8"],"Date":["Wed, 11 Dec 2019 20:35:33 GMT"],"Etag":["W/\"3c810-16e8d73c970\""],"Last-Modified":["Thu, 21 Nov 2019 10:13:26 GMT"],"Server":["Cowboy"],"Transfer-Encoding":["chunked"],"Vary":["Accept-Encoding"],"Via":["1.1 vegur, 1.1 vegur"]},"statusCode":200,"statusLine":"HTTP/1.1 200 OK","timestamp":1576096534024.7532,"url":"https://www.balena.io/etcher/success-banner/?etcher-version=1.5.52&api-version=2&optOutAnalytics=false"},"applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:34 GMT+0100 (Mitteleuropäische Zeit) Open modal ({"name":"drive-selector","applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:34 GMT+0100 (Mitteleuropäische Zeit) Reselect drive ({"applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:35 GMT+0100 (Mitteleuropäische Zeit) Toggle drive ({"drive":{"size":7932477440,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive10","error":null,"isReadOnly":false,"displayName":"\\\\.\\PhysicalDrive10","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive10","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[],"description":"Generic STORAGE DEVICE USB Device","isUAS":false},"previouslySelected":false,"applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:36 GMT+0100 (Mitteleuropäische Zeit) Modal accepted ({"name":"drive-selector","applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:38 GMT+0100 (Mitteleuropäische Zeit) Open image selector ({"applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:41 GMT+0100 (Mitteleuropäische Zeit) Select image ({"image":{"size":7654604800,"name":"pglu_raspbian-os_v3.3_8GB.img","hasMBR":true,"partitions":[{"offset":32256,"size":66060288,"type":6,"index":1},{"offset":74027520,"size":7340032000,"type":131,"index":2}],"path":"D:\\Rolf\\Dropbox\\01_-_PGLU\\999-PGLU_MASTER\\pglu_raspbian-os_v3.3_8GB.img","extension":"img","logo":false,"blockMap":false},"applicationSessionUuid":"2ef7ca75-9d69-4c0a-b111-4d61859c3d9d","flashingWorkflowUuid":"8802aece-75fa-475b-808b-10e7a2227e4c","sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:35:42 GMT+0100 (Mitteleuropäische Zeit) Flashing 0 devices, 0% at null MB/s (total null MB/s) eta in undefineds with 0 failed devices
[... cut ...] ,"devicePath":null,"mountpoints":[],"description":"Generic STORAGE DEVICE USB Device","isUAS":false}],"driveCount":2,"uuid":"893970b0-5b8f-4509-bd8d-f84aa062e0cf","unmountOnSuccess":true,"validateWriteOnSuccess":false,"trim":false,"sample":0.1})
C:\Users\Rolf\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Dec 11 2019 21:36:06 GMT+0100 (Mitteleuropäische Zeit) Flashing 1 devices, 0% at 2.1 MB/s (total 2.1 MB/s) eta in 3649s with 1 failed devices

lurch, 2019-12-12:
@pglu Have you verified that Etcher is able to successfully write both cards individually, to rule out any possible hardware problems?

closed 2023-07-12 by dfunckt