balena-io/etcher #3013: balenaEtcher don't show full names of file and card reader. Instead the dots are show

bessarabov, 2019-12-14:
- **Etcher version:**  — 1.5.56
- **Operating system and architecture:**  macOS 10.14.6 (18G95)
- **Image flashed:** `2019-09-26-raspbian-buster-lite.zip`
- **Do you see any meaningful error information in the DevTools?**  — no

First of all thank you for the great tool. I occasionally use it and it works great, but there is a tiny thing that makes me sad. The file name and the card reader name are not fully shown, only the beginning and end are visible (and there are dots in between).

I would be happy if this soft don't hide part of the names but show them fully without change.

[image]
[image]
[image]

lurch, 2019-12-16:
The names can be **very** long, which IIRC is why the '...' is used so they fit okay in the GUI. What's wrong with using the hover-tooltips, as your screenshots show?

bessarabov, 2019-12-20:
Yes, the names can be very long, and it is ok to strip the length of the very long names.

But the names in my example are not long names. They are just ordinary names:

 * `2019-09-26-raspbian-buster-lite.zip` (35 chars)
 * `APPLE SD Card Reader Media` (26 chars)

> What's wrong with using the hover-tooltips, as your screenshots show?

This is just a bad user experience, it looks ugly and it force the user to do unnecessary actions just to make sure the selection is correct.

lurch, 2019-12-21:
I suspect this is one of those situations where a "long name" for one person is a "normal name" for another person? :man_shrugging: 

IIRC this is the bit that adds the dots in the middle https://github.com/balena-io/etcher/search?q=middleellipsis&type=Code