balena-io/etcher #3014: Duplicate version in version window

bessarabov, 2019-12-14:
- **Etcher version:**  — 1.5.56
- **Operating system and architecture:**  macOS 10.14.6 (18G95)
- **Image flashed:** `2019-09-26-raspbian-buster-lite.zip`
- **Do you see any meaningful error information in the DevTools?**  — no

In version window the version is written 2 times. Maybe it is better to write it only once?

[image]

shawaj, 2020-12-13:
@bessarabov do you still get this issue in v1.5.112?

fantomnotabene, 2021-02-27:
@shawaj [image]

lurch, 2021-02-27:
_Possibly_ related to https://github.com/balena-io/etcher/blob/master/lib/gui/menu.ts#L94 and https://www.electronjs.org/docs/api/menu-item#roles :shrug: