balena-io/etcher #3015: What's the diff between etcher and dd-like tools?

hongyi-zhao, 2019-12-15:
Hi,

I just want to know what's diff when doing image clone to disk with etcher and dd-like tools?

Any hints/notes for this?

Regards

lurch, 2019-12-16:
IMHO:
1) much nicer UI (which is the same across all OSes)
2) verification of the written data
3) support for writing multiple drives at once
4) see https://www.balena.io/blog/tag/etcher/ for more

But at a fundamental level, they both do the same thing of copying a source-image to a destination-drive, in a byte-perfect manner (if that was what you wanted to know?). They both overwrite whatever was on the drive previously.

hongyi-zhao, 2019-12-16:
Thanks for your notes.

Consider the following:

The dd-like tools ( say, dd, ddrescue, partclone.dd, and so on. ) can use one of the partition as the destition instead on the whole disk:

$ dd if=some.img of=/dev/sdc1 bs=1M

How ether for this type of job?

lurch, 2019-12-16:
AFAIK Etcher can't do that.
See also #413