balena-io/etcher #3016: v1.5.70 has no drag n drop feature - is this intended?

2play, 2019-12-15:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

ideologysec, 2019-12-19:
more details are very helpful for reproducing.

What OS flavor Mac? Linux? Win10? what version?

What version of balenaEtcher?

2play, 2019-12-19:
hi @ideologysec  Eric
Windows10. If you set it to run as administrator the drag drop feature does not work to load image file
if as normal works ok but the before burning its asking cmd to run at higher privileges to format and start burning.
I wanted to avoid this (which works) if run as administrator but drag and drop feature does not work which is a great advantage of Balena etcher :-)

2play, 2019-12-19:
btw i tried to edit the title as this applies to older vesion 69 when tried to test

lurch, 2019-12-19:
Is there an older version at https://github.com/balena-io/etcher/releases which "works properly", so the Etcher developers can identify whether this is a regression?
Maybe the title should be renamed to something like "drag n drop doesn't work when running as administrator", since you seem to be saying drag'n'drop _does_ work when you're running as a normal user? :man_shrugging:

2play, 2019-12-19:
@lurch I will check but not tested. I started using as admin this week and initially thought it was broken on the latest version. Then tried 69 witth and wihtout and got same. 
so it something to be checked if was initally like that

2play, 2019-12-19:
ok I tested all the way down to balenaEtcher-Portable-1.5.15-x64.exe
no admin: works ok
admin: not working

Aceto1, 2020-01-14:
This is probably a Windows issue. Microsoft article here: https://docs.microsoft.com/de-de/archive/blogs/patricka/q-why-doesnt-drag-and-drop-work-when-my-application-is-running-elevated-a-mandatory-integrity-control-and-uipi

2play, 2020-01-14:
@Aceto1 as per article  you posted...
"The best solution is to only use drag and drop between the same MIC levels. With UAC enabled, Windows Explorer will run at a medium MIC level.  Therefore, your application ... needs to run at medium (or lower) MIC level.  The bottom line is that drag and drop from Windows Explorer will not work if your application is elevated.  
**If you find yourself in this situation, you may need to rethink your application design or not support drag-and-drop with UAC enabled.**"

ideologysec, 2020-01-14:
Is there a reason you're running Etcher as administrator in the first place as opposed to a normal user?

2play, 2020-01-17:
@ideologysec I tried the elevated yes to avoid being asked for accepting the cmd part before start burning

ideologysec, 2020-01-23:
This is probably a wontfix, honestly.

I'm not on the Etcher team, but this appears working as designed. Etcher can't change the Windows security model, and if you're dragging and dropping from Explorer running as a normal user to Etcher running as an admin, Windows won't allow that to cross the security boundary.

Run it as a normal user and accept the admin prompt before burning, or spawn an explorer window as admin to be able to drag and drop.

2play, 2020-01-24:
Probably. I have my UAC enabled i frag drop and when etcher asks to erase/write I authorize the cmd UAC
It would be nice to be smoother but I guess I ll have to live with it

2play, 2020-02-12:
Even with UAC disabled the drag drop doesnt work
If I could get and install the publisher certificate locally on my pc as trusted then might work.
But cert cant be downloaded.
Just an update for latest v1.75