balena-io/etcher #3018: Contribute on windows

cedricve, 2019-12-19:
- **Etcher version:**  1.5
- **Operating system and architecture:** windows 10

Hey I'm trying to build the project on Windows (I already managed to make it work on OSX).
But even after installing all the software + plugins, I still get the `make not found` issue?
https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md

Any idea?

[image]

cedricve, 2019-12-20:
I tried downloading a different sha256sum.exe but installation throws errors. Anyone who can share with me if the tutorial is up to date?

Cédric

cedricve, 2019-12-21:
it was a hell of a ride, but I got it working in the end. my conclusion is that the Contributing section is lacking a lot of details. PM me if you need more info.

closed 2019-12-21 by cedricve

lurch, 2019-12-22:
AFAICT (this is only an educated guess) the Etcher build is now controlled by https://github.com/balena-io/etcher/blob/master/.resinci.json but I've no idea if any of the ResinCI code / process is open source? :man_shrugging:

thundron, 2019-12-24:
@cedricve First, thank you a lot for persisting and getting it to work, it really means a lot to us.
Would you mind sharing your findings on which packages are missing, which binaries and the other details? The CI code isn't totally open source (as in, the main structure is not), but the contributing guidelines aim to ease development for whoever wants to contribute to this repository and thus, the official version of Etcher.
Any help is highly appreciated be it in the docs or for other functionalities, for which you can also make a PR for so we can "just" review the PR instead of waiting for us to fix the docs, which in this case isn't trivial unfortunately as we made a lot of changes in various places (as I'm sure you could tell by following the guide)