balena-io/etcher #3019: Etcher - brand new USB RAW, unusable

rand538, 2019-12-21:
- Etcher version: v1.5.70
- Operating system and architecture: Win 10 Pro 1909 / i3-5005U
- Image flashed: Debian Live 10.2.0

I flashed Debian Live iso on USB I bought hours before that. It worked, but I didn’t want to keep Debian so I immediately rebooted. In Win 10 Pro I tried to format, but ‘Windows was unable to complete format’ pops up.
In cmd(admin), I tried to clean with diskpart, but it would give me error: Access is denied. USB disappears from both File and Disk Management. DM also couldn’t format.
I flash another USB with Gparted.iso and format USB to FAT32. Now it shows in Win, but keeps disappearing. In format in fdisk. Than I manage to format in Windows and it appears to be fine. But doing anything with it (eg. writing files) disconnects it and it shows it as RAW, Healthy. Sometimes reverts to Unallocated.
Easeus showed no bad sectors.

So most of the solutions I found don’t work:

1. Diskpart and clean
2. Format in File / Disk Management
3. Format in gdisk or fdisk
4. Using Easeus, AOMEI

lurch, 2019-12-21:
Just out of curiosity, is Etcher itself still able to flash a different image to the drive?

dylanf, 2019-12-21:
I am experiencing a similar problem but with SDHC cards, etcher v1.5.70
Unfortunately I cannot determine exactly if the problem lies with etcher, the (old) raspberry Pi that I then tried to boot from the cards, or the new card reader that I am using to write them.

But the end result is I have 2 inaccessible sdhc cards in the last couple of hours, one of them brand new.  Note I am running in windows 10 (which can no longer even identify the cards).

rand538, 2019-12-22:
I solved the problem. Whenever I went to diskpart to clean and format, I would add 'quick' attribute. I tried formatting without the quick part and it worked. It took longer but now my USB is fine. Tried copying files and reading them, no problem. Before this, some methods would appear to be successful, but eventually the USB would revert to RAW. Now it's stable. 
Back to Rufus, Etcher never again.

lurch, 2019-12-23:
As mentioned in many other issues, this isn't a "problem" of Etcher, but a "problem" of the image you're tying to flash. See e.g. https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos for more info.

closed 2019-12-24 by thundron