balena-io/etcher #3022: etcher on os x allowes no write (v1.5.70)

pwy300, 2019-12-22:
- Etcher version: v1.5.70
- Operating system and architecture: os x 
- Image flashed: all

lurch, 2019-12-22:
I _suspect_ the Etcher developers would appreciate more detail (e.g. a screenshot and/or any error messages), in order to try and determine what might be going wrong? :wink:

thundron, 2019-12-24:
More details would definitely be appreciated, otherwise there's nothing we can do.
Could you tell us a bit more about the behaviour, as well as filling in the last point of the issue template (logs in the console after the failed operation with `cmd + opt + i`)?

rememberedbound, 2019-12-30:
This fixed it for me, I had a previous FTDI USB serial driver installed, on Catalina it grabs write access to the USB serial device in exclusive mode and won't allow anything else to talk to it. Installed 2.3 from here https://www.ftdichip.com/Drivers/VCP.htm and that fixed it without even restarting etcher.

No having to use sudo etc either.

closed 2020-01-02 by thundron