balena-io/etcher #3027: ONN micro SDHC card failing to flash

GalexyEndX, 2020-01-04:
- Etcher Version: 1.5.70
- OS: MacOS Catalina 10.15.3 Beta (19D49f)
- Image: [chromiumos_test_image_r77r2-rpi3b.img](https://github.com/FydeOS/chromium_os_for_raspberry_pi/releases)
- Flashing To: 32GB ONN Class 10 Micro SDHC Card
- Error:
`Fri Jan 03 2020 18:44:13 GMT-0700 (Mountain Standard Time) Done ({"errors":[{"name":"Error","code":"EIO","syscall":"write","errno":-5,"device":"/dev/disk2"}],"devices":{"failed":1,"successful":0},"status":"started","image":"/Users/bensova/Desktop/chromiumos_test_image_r77r2-rpi3b.img","drives":[{"size":31312576512,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"Secure Digital","isCard":true,"isUSB":false,"devicePath":"IODeviceTree:/PCI0@0/RP01@1C/SDXC@0,1","mountpoints":[],"description":"Apple SDXC Reader Media","isUAS":null}],"driveCount":1,"uuid":"0e3582e0-3a28-4dcc-addc-a3a812209daa","flashInstanceUuid":"0e3582e0-3a28-4dcc-addc-a3a812209daa","unmountOnSuccess":true,"validateWriteOnSuccess":true,"trim":false,"applicationSessionUuid":"000da753-599a-4a17-84a0-e462057e0f26","flashingWorkflowUuid":"6b2c036a-7f99-404d-be14-2cb67abb7c32","sample":0.1})
`
Note: After each attempt to flash, I have to run the card through Disk Utility around 5 times (the other ~4 times I get a fail form DiskUtil) to get it back to the same point.

lurch, 2020-01-04:
> I have to run the card through Disk Utility around 5 times (the other ~4 times I get a fail form DiskUtil)

Doesn't that suggest that the card itself is faulty? :confused:

GalexyEndX, 2020-01-04:
Probably. However, I was able to get it to run [NOOBS Lite](https://www.raspberrypi.org/downloads/noobs/). It was a .zip and I took the files within it and put it on the card, that seemed to work.

lurch, 2020-01-05:
NOOBS Lite is very small, so perhaps it's small enough to fit within the 'working' section of you SD cad? :man_shrugging: 

If you want to do a thorough check of your SD card, have a look at https://fight-flash-fraud.readthedocs.io/en/latest/introduction.html  (and if your SD card **is** faulty, hopefully it's new enough that you're able to return it under warranty).

closed 2020-04-14 by GalexyEndX