balena-io/etcher #3028: with 2nd external Screen - Etcher started but no Window rendered!

phenomeus, 2020-01-04:
- **Etcher version:** 
Version 1.5.70 (1.5.70)
- **Operating system and architecture:** 
macOS 10.15.2
- **Image flashed:** 
not able
- **Do you see any meaningful error information in the DevTools?** 
no

if I have my 34" external LG Screen attached with Thunderbolt / Displayport to my MacBook Pro with the latest OS, etcher is started but not Window is actually rendered. Only when I disconnect the cable its instantly visible. or if I disconnect the cable and restart etcher, its instantly usable. 

but never with the 2nd screen attached.

phenomeus, 2020-01-17:
any idea? easy to reproduce

thundron, 2020-01-20:
@phenomeus Could it be that it's started in a secondary desktop? I often use two screens while developing and never had an issue.
Sometimes it can be slow to start, so you can also try to give it some more time before unplugging

phenomeus, 2020-01-21:
@thundron yeah you are right, I had to wait more than a minute to see the start screen and while using its horrible slow in the reaction… you can see how the its "thinking" xD 

click - wait - select - wait - click - wait - select -wait…

lurch, 2020-01-21:
> its horrible slow in the reaction… you can see how the its "thinking" xD

Hmm, another example of #2594 or #2740 or #2770 ?

phenomeus, 2020-01-22:
seems likely

closed 2020-11-24 by thundron