balena-io/etcher #3029: Installation from repository on Ubuntu 18.04 fails with "failed to create symbolic link"

fantauzzi, 2020-01-09:
- **Etcher version:** 
1.5.51
- **Operating system and architecture:** 
`~> uname -a`
`Linux Ono-Sendai7 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
- **Image flashed:** 
N/A
- **Do you see any meaningful error information in the DevTools?** 
N/A

Added the following repository:
`deb https://deb.etcher.io stable etcher`

Here the shell transcript, with the error

```
~> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.RIUD5VuxYc/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: key 379CE192D401AB61: public key "Bintray (by JFrog) <bintray@bintray.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
~> sudo apt update
[...]
~> sudo apt install balena-etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  geany-common linux-headers-4.15.0-72 linux-headers-4.15.0-72-generic
  linux-image-4.15.0-72-generic linux-modules-4.15.0-72-generic
  linux-modules-extra-4.15.0-72-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  gconf-service gconf-service-backend gconf2 gconf2-common libappindicator1
  libgconf-2-4 libindicator7 libpango1.0-0 libpangox-1.0-0
Suggested packages:
  gconf-defaults-service indicator-application
The following NEW packages will be installed:
  balena-etcher-electron gconf-service gconf-service-backend gconf2
  gconf2-common libappindicator1 libgconf-2-4 libindicator7 libpango1.0-0
  libpangox-1.0-0
0 upgraded, 10 newly installed, 0 to remove and 2 not upgraded.
Need to get 58,2 MB of archives.
After this operation, 238 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:2 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 libindicator7 amd64 16.10.0+18.04.20180321.1-0ubuntu1 [21,7 kB]
Get:1 https://dl.bintray.com/etcher/debian stable/etcher amd64 balena-etcher-electron amd64 1.5.51 [57,2 MB]
Get:3 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 libappindicator1 amd64 12.10.1+18.04.20180322.1-0ubuntu1 [19,4 kB]
Get:4 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 libpangox-1.0-0 amd64 0.0.2-5 [41,7 kB]
Get:5 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf2-common all 3.2.6-4ubuntu1 [700 kB]
Get:6 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 libgconf-2-4 amd64 3.2.6-4ubuntu1 [84,8 kB]
Get:7 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf-service-backend amd64 3.2.6-4ubuntu1 [58,1 kB]
Get:8 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf-service amd64 3.2.6-4ubuntu1 [2.036 B]
Get:9 http://it.archive.ubuntu.com/ubuntu bionic/universe amd64 gconf2 amd64 3.2.6-4ubuntu1 [66,9 kB]
Get:10 http://it.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 libpango1.0-0 amd64 1.40.14-1ubuntu0.1 [3.344 B]
Fetched 58,2 MB in 5s (11,0 MB/s)                                              
Selecting previously unselected package libindicator7.
(Reading database ... 342425 files and directories currently installed.)
Preparing to unpack .../0-libindicator7_16.10.0+18.04.20180321.1-0ubuntu1_amd64.deb ...
Unpacking libindicator7 (16.10.0+18.04.20180321.1-0ubuntu1) ...
Selecting previously unselected package libappindicator1.
Preparing to unpack .../1-libappindicator1_12.10.1+18.04.20180322.1-0ubuntu1_amd64.deb ...
Unpacking libappindicator1 (12.10.1+18.04.20180322.1-0ubuntu1) ...
Selecting previously unselected package libpangox-1.0-0:amd64.
Preparing to unpack .../2-libpangox-1.0-0_0.0.2-5_amd64.deb ...
Unpacking libpangox-1.0-0:amd64 (0.0.2-5) ...
Selecting previously unselected package gconf2-common.
Preparing to unpack .../3-gconf2-common_3.2.6-4ubuntu1_all.deb ...
Unpacking gconf2-common (3.2.6-4ubuntu1) ...
Selecting previously unselected package libgconf-2-4:amd64.
Preparing to unpack .../4-libgconf-2-4_3.2.6-4ubuntu1_amd64.deb ...
Unpacking libgconf-2-4:amd64 (3.2.6-4ubuntu1) ...
Selecting previously unselected package gconf-service-backend.
Preparing to unpack .../5-gconf-service-backend_3.2.6-4ubuntu1_amd64.deb ...
Unpacking gconf-service-backend (3.2.6-4ubuntu1) ...
Selecting previously unselected package gconf-service.
Preparing to unpack .../6-gconf-service_3.2.6-4ubuntu1_amd64.deb ...
Unpacking gconf-service (3.2.6-4ubuntu1) ...
Selecting previously unselected package gconf2.
Preparing to unpack .../7-gconf2_3.2.6-4ubuntu1_amd64.deb ...
Unpacking gconf2 (3.2.6-4ubuntu1) ...
Selecting previously unselected package libpango1.0-0:amd64.
Preparing to unpack .../8-libpango1.0-0_1.40.14-1ubuntu0.1_amd64.deb ...
Unpacking libpango1.0-0:amd64 (1.40.14-1ubuntu0.1) ...
Selecting previously unselected package balena-etcher-electron.
Preparing to unpack .../9-balena-etcher-electron_1.5.51_amd64.deb ...
Unpacking balena-etcher-electron (1.5.51) ...
Setting up gconf2-common (3.2.6-4ubuntu1) ...

Creating config file /etc/gconf/2/path with new version
Setting up libindicator7 (16.10.0+18.04.20180321.1-0ubuntu1) ...
Setting up libappindicator1 (12.10.1+18.04.20180322.1-0ubuntu1) ...
Setting up libpangox-1.0-0:amd64 (0.0.2-5) ...
Setting up libpango1.0-0:amd64 (1.40.14-1ubuntu0.1) ...
Processing triggers for sgml-base (1.29) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2) ...
Setting up libgconf-2-4:amd64 (3.2.6-4ubuntu1) ...
Setting up gconf-service-backend (3.2.6-4ubuntu1) ...
Setting up gconf-service (3.2.6-4ubuntu1) ...
Setting up gconf2 (3.2.6-4ubuntu1) ...
Setting up balena-etcher-electron (1.5.51) ...
ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory
dpkg: error processing package balena-etcher-electron (--configure):
 installed balena-etcher-electron package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Errors were encountered while processing:
 balena-etcher-electron
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

lurch, 2020-01-09:
Seems to be a duplicate of #2458 (I guess this issue also happening with `balena-etcher-electron_1.5.51_amd64.deb` probably means that previous issue was closed prematurely?)

See my comment in #2458 for my guess of what might be going wrong.

fantauzzi, 2020-01-09:
Manually creating the required directory allowed installation to complete successfully:
`sudo mkdir /usr/local/bin
`

lurch, 2020-01-09:
IMHO this should be re-opened - just because there's a manual workaround, doesn't mean it isn't a bug in the .deb packaging :wink:

fantauzzi, 2020-01-10:
Re-opening the issue as per another user suggestion, as I provided only a workaround.