balena-io/etcher #3030: I cant flash SD Card

Daniella-Robotics, 2020-01-09:
- **Etcher version:**  balenaEtcher-Portable-1.5.70
- **Operating system and architecture:** Windows 10
- **Image flashed:** No
- **Do you see any meaningful error information in the DevTools?** The writer process ended unexpectedly...

Please help me get through this stage.  I follow the instructions correctly but just can't get it to flash the sd card. it flashes till 99 percent then give me the error message. how do i fix this?

lurch, 2020-01-09:
Seems to be many other people having the same problem
https://github.com/balena-io/etcher/search?q=The+writer+process+ended+unexpectedly&type=Issues

ning9ng, 2020-01-15:
Running Etcher in Administrator mode solved the Write Issue for me on a Kingstone SHDC card.

MoJeeza, 2020-01-21:
I have the same problem - I can format the card both Quick and complete format - but can't flash the image using Etcher V 1.5.71 on  Windows 10 (latest patch) to a SanDisk 64Gb microSD Card.
[etcher.pdf](https://github.com/balena-io/etcher/files/4093618/etcher.pdf)

MoJeeza, 2020-01-21:
Sorry, team...I just found the solution that worked for me...

The problem is with "controlled folder access". Once I allowed the "balenaEtcher.exe" access to "restricted folder areas" I am fine. Image flashed successfully.

It was just by co-incidence that I looked at the messages in this "notification" area.

lurch, 2020-01-22:
@thundron Maybe [controlled folder access](https://github.com/balena-io/etcher/search?q=%22controlled+folder+access%22&type=Issues) should be mentioned in the Etcher documentation? :man_shrugging:

MoJeeza, 2020-01-22:
Has been working perfectly since making that change..

mischief957, 2020-04-10:
disableing my controlled folder access in the windows defender fixed my issue.

settings-security at a glance-virus and threat protection-manage settings- controlled folder access-manage controlled folder access- off

that should fix the issue, if not then try runnining as admin

closed 2023-07-12 by dfunckt