balena-io/etcher #3031: Stuck on "Starting"

etricky, 2020-01-13:
- **Etcher version:** 1.5.57
- **Operating system and architecture:** Windows 10 64bits
- **Image flashed:** D:\Downloads\Software\hassio\hassos_rpi3-3.7.img.gz
- **Do you see any meaningful error information in the DevTools?** No

Tried to flash the image to an 64GB SD card but got stuck on Starting and no errors or warnings in the log. Tried to run Etcher as an Administrator with the same result.

[-1578873400126.log](https://github.com/balena-io/etcher/files/4051773/-1578873400126.log)

lurch, 2020-01-16:
Etcher sometimes has "issues" with large gzip (`.gz`) compressed images. Do you get any different behaviour if you uncompress the file first and try flashing `hassos_rpi3-3.7.img` instead?

etricky, 2020-01-17:
It doesn't matter if I use the gz or img file that is inside the gz file.
I managed to fix this by formatting the sd card and erasing the existent
partitions but I would expect some kind of feedback of the application
instead of just getting stuck on starting.

discojohn99, 2020-04-21:
- Etcher version: 1.5.81
- Operating system and architecture: Windows 10 64bits
- Image flashed: C:\Users\jve\Downloads\retropie-4.5.1-rpi2_rpi3 (1).img
- Do you see any meaningful error information in the DevTools? Not sure what to look for...

Using a 32Gb Samsung MicroSD out the box yesterday.  

Etcher sees the image and the SD card, "Starting" shown but never gets off 0%.  I have opened Etcher as Admin.  Any guidance v welcome!!

DrakeProvost, 2020-04-25:
A similar problem as @discojohn99 is having.
- **Etcher version:** 1.5.81
- **OS and architecture:** Windows 10 64bit
- **Image flashed:** C:\Users\mddmp\Downloads\ubuntu-18.04.4-preinstalled-server-arm64+raspi3.img.xz
- **Do you see any meaningful error information in the DevTools?**: Not sure what to look for

Using a brand-new 32GB SanDisk Extreme Pro microSDHC UHS-I card.

Etcher sees the image and SD card, but gets stuck at "starting" without even a progress bar showing. Opening as admin doesn't help.

thundron, 2020-09-28:
Could you please refer to this [forum post](https://forums.balena.io/t/cannot-create-boot-media/189673) and get back here with the results on your system trying to replicate the steps we suggested to that user?

closed 2020-09-28 by thundron