balena-io/etcher #3033: Could not install packages due to an EnvironmentError MacOS Catalina

JSReds, 2020-01-15:
- **Operating system and architecture: Mac OS Catalina v10.15.2** 

following Etcher's development guide, running the command `sudo pip install -r requirements.txt` I get the following error:

```
Could not install packages due to an EnvironmentError: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/parser.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/parser.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/parser.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/relativedelta.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/relativedelta.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/relativedelta.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo/zoneinfo-2010g.tar.gz', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo/zoneinfo-2010g.tar.gz', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo/zoneinfo-2010g.tar.gz'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo/__init__.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo/__init__.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo/__init__.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo/__init__.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/zoneinfo', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/zoneinfo'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tz.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tz.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tz.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/relativedelta.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/relativedelta.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/relativedelta.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/rrule.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/rrule.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/rrule.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/__init__.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/__init__.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/parser.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/parser.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/parser.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tzwin.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tzwin.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tzwin.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/rrule.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/rrule.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/rrule.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/__init__.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/__init__.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/easter.py', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/easter.py', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/easter.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/easter.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/easter.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/easter.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tz.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tz.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tz.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil/tzwin.pyc', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tzwin.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ/tzwin.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/dateutil', '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ', "[Errno 1] Operation not permitted: '/private/var/folders/t2/7hg92fh96ljbpzrvzv6nvtjh0000gn/T/pip-uninstall-P0P6AZ'")]
```

To solve the problem on MacOS Catalina, in my case I just added the `--user` flag:

`sudo pip install -r requirements.txt --user`

thundron, 2020-01-15:
Did you try without sudo so just `pip install -r requirements.txt --user` by chance?

JSReds, 2020-01-15:
yes I did but then the output is: 

```ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/Users/${whoami}/Library/Python/2.7'
Check the permissions.```

lurch, 2020-01-16:
> `Permission denied: '/Users/${whoami}/Library/Python/2.7' Check the permissions.`

That's _probably_ because you ran `sudo pip install -r requirements.txt --user` first, which created `'/Users/${whoami}/Library/Python/2.7'` as a root-owned directory; and so when you later ran `pip install -r requirements.txt --user` (without sudo) it couldn't overwrite the root-owned files?

I **think** if you did a `sudo chown -R ${whoami} /Users/${whoami}/Library/Python/2.7` then that _should_ allow a `pip install -r requirements.txt --user` (without sudo) to run successfully?

Disclaimer: I'm a Linux user, I don't have a Mac :wink: