balena-io/etcher #3037: Running Etcher from the source code

Yutyo, 2020-01-19:
Hi, I have cloned this repo and performed `npm install` but when I performed `npm start` , it gave error during . electron command:
`Error launching app
Unable to find Electron app at /home/yutyo/git/etcher

Cannot find module '/home/yutyo/git/etcher/generated/etcher.js'. Please verify that the package.json has a valid "main" entry
`
Are there any other way to run Etcher from the source code (this repo)?

lurch, 2020-01-20:
Have you read https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md ?

thundron, 2020-01-20:
> Although I've not personally checked if the instructions there still work.

They do on macOS/Linux, while we need some changes for the Windows section.

@Yutyo Check the mentioned document, if you find any issues there you can re-open 👍

closed 2020-01-20 by thundron

bpp9614, 2020-04-08:
They need lot of changes for Windows section.