balena-io/etcher #3039: 1.5.71 fails

delldallas, 2020-01-20:
- **1.5.71** 
- **WIN7-64bit** 
- **DietPi_RPi-ARMv6-Buster.img** 
- **Do you see any meaningful error information in the DevTools?
how do i get Ctrl+Alt+I to work?** 

the version number used to be on the main screen on OLDer version in LowerLeft corner.
I works with really old 1.4.4 so I will keep it.

thundron, 2020-01-20:
You can access DevTools with `ctrl + shift + i` on Windows, shall you need to, but the version number is available in the settings.

closed 2020-01-20 by thundron

delldallas, 2020-02-01:
newer 1.5.73 fails also WIN7  ctrl+shift+i  says:
[{"name":"Error","device":"\\\\.\\PhysicalDrive1"}],"devices":{"failed":1,"successful":0},"sample":0.1})
3C:\Users\SCOTT\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 [withTheme] You are not using a ThemeProvider nor passing a theme prop or a theme in defaultProps in component class "Component"
console.<computed> @ C:\Users\SCOTT\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58
/C:/Users/SCOTT/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png:1 Failed to load resource: net::ERR_FILE_NOT_FOUND

my  /app.asar is a 150 MB file ,  is it a  library, 
/C:/Users/SCOTT/AppData/Local/Programs/balena-etcher/resources/app.asar  
 NOT a directory  TO ../app.asar/lib.assets/icon.png

1.4.4  WORKS but   ctrl+shift+i   says: 
{"stack":"TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined\n    at Object.exports.notify (file:///C:/Users/DELLDALLAS/AppData/Local/Programs/etcher/resources/app.asar/generated/gui.js:118076:55)\n    at file:///C:/Users/SCOTT/AppData/Local/Programs/etcher/resources/app.asar/generated/gui.js:65181:27\n    at tryCatcher 

hope this helps.

thundron, 2020-02-12:
@delldallas Can you try with the recent 1.5.76? Are you able to flash or does it fail before the end with an error? If so, what does the error say?

lurch, 2020-02-12:
@delldallas `.asar` files are archives used by electron https://github.com/electron/asar

delldallas, 2020-02-18:
Running Windows7   1.5.77

missing ﻿file:///C:/Users/SCOTT/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png
file:///C:/Users/SCOTT/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png
app.asar is a 104MB file not a SUBDIR
Did the directory structure change?  Maybe a remove/reinstall will fix the WINDOWS7 version

https://www.balena.io/etcher/success-banner/?etcher-version=1.5.76&api-version=2   <- fix 1.5.77
file:///C:/Users/SCOTT/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png

lurch, 2020-02-18:
@delldallas It seems very odd that you're getting the same error message on both 1.5.73 and 1.5.77 - are you sure you're not accidentally still running the old version? :man_shrugging: