balena-io/etcher #3040: UI almost completely unresponsive after selecting source image

pistonet, 2020-01-24:
- **Etcher version:** v1.5.71 (x64, AppImage)
- **Operating system and architecture:** Kubuntu 18.04.3 LTS (x86_64)
- **Image flashed:** ABITTI2003R (227) (ktp-etcher.zip)
- **Do you see any meaningful error information in the DevTools?** Probably, there's a warning.

After loading the image clicking basically anything on the UI – whether it's choosing a new image, choosing the target device or clicking the settings – causes the program to freeze for something around 10-20 seconds. Usually the fans on my laptop ramp up too.

This feels like something similar to #2740 but since I have issues only after selecting an image I'm submitting this as a new issue. I might be wrong though...

The image I'm using is part of the Finnish Matriculation Examination Board's digital exam system, it's available [on this page](https://www.abitti.fi/fi/paivitykset/) at least until they update to a newer version, the link is 'koetilan palvelimen levynkuva'. I have the same issue with older versions of their image, while something like GParted's gparted-live-1.0.0-5-i686.iso doesn't cause this.

This makes me think the issue might be their fault, something related to how they made the disk image, but the image works fine on older versions of Etcher...

The interesting part is that Etcher used to work fine, and I thought it was something I updated on my computer that broke it, but actually it was the etcher auto updater doing it's thing. So I tried some older versions and here's what I got from the versions I tested...
I have this problem on: v1.5.71, v1.5.70
Works fine: v1.5.69, v1.5.68, v1.5.64

The error I'm getting on v1.5.71 is here:
[image]

Also the logfile of this same error is here:
[-1579857777677.log](https://github.com/balena-io/etcher/files/4107809/-1579857777677.log)
It's the "Invalid prop" error on line 16 I think.

Thanks.

mtreinik, 2020-01-28:
I can reproduce this. My Etcher halts for tens of seconds when I try to open the etcher zip file http://static.abitti.fi/etcher-usb/ktp-etcher.zip

I did some experimentation and Etcher seems to be choking on the .meta/logo.svg included.

The logo is a 500x500 pixel SVG image that is 415729 bytes. I know that there have not been recent changes to that image.

Maybe something has changed in the way that Etcher processes the logo image?

mtreinik, 2020-01-29:
@pistonet, we fixed the problem with the particular etcher zip at http://static.abitti.fi/etcher-usb/ktp-etcher.zip by replacing logo.svg by a more lightweight image file.

In case Balena wants to improve image loading in Etcher, here is the old image file that halts Etcher for a long time:
https://www.dropbox.com/s/hshk3yuaosat45d/logo-old.svg?dl=0

Here is the new image that works with no delay:
https://www.dropbox.com/s/c5nv3pnmlump6tk/logo-new.svg?dl=0

pistonet, 2020-01-30:
The new image on http://static.abitti.fi/etcher-usb/ktp-etcher.zip works now.
@mtreinik you have my thanks!

I should add for anyone having this issue with another zip file:
In my case removing .meta/logo.svg and rezipping the file without it obviously also removes the problem, so there's a temporary fix for this issue.

Glad this seems to be working now, I hope this gets fixed completely in a later update.