balena-io/etcher #3049: Display warning message to user

Moltkex, 2020-02-02:
Could you make etcher display a warning message to users so they know every piece of data in the device they're about to flash will be lost forever? I just recommended etcher to someone to flash a Linux OS and it worked, only he forgot to take the data  in the USB out before so he lost it, I agree he should've been more careful but other tools do WARN users that whatever data is in the device will be destroyed so they make sure there's nothing on it before proceeding. I think it's a nice and useful feature that'll make your software even better than it already is.  

Thanks.

lurch, 2020-02-03:
LOL, #2638 is asking for fewer warning dialogs... :wink:

Moltkex, 2020-02-03:
@lurch fewer warnings? I've seen none and it looks like #2638 is one of a kind since it's about warning about some devices being larger than X size, which is entirely different from displaying a simple warning when you click on proceed/flash/continue: **Warning: Every single piece of data in the device will be lost forever, are you Ok with that?** Or some similar message, the point is, some distracted none-techy-savy, unfamiliar people with these kind of procedures can save their data before flashing their devices. I know if am to create a Live USB everything in the drive will be deleted so I save any data on the device somewhere else safe before proceeding, and even then I'd still appreciate if such a warning message were to be displayed and I imagine those who have no idea this will happen would as well.

Guiorgy, 2020-03-11:
The best middle ground in my opinion, is to have such a message and have a checkbox with something like "don't show again", maybe even checked by default.

dfunckt, 2023-07-12:
Closing in favor of #3289

closed 2023-07-12 by dfunckt