balena-io/etcher #3050: The program stopped working, before that everything worked fine

Sveshlex, 2020-02-02:
[https://i.imgur.com/EkAwDZR.png](url)

hkyq, 2020-02-02:
Looks like you are on Linux, running from terminal with `etcher` should produce a more detailed message.

Sveshlex, 2020-02-03:
it turns out exactly the same error

thundron, 2020-02-11:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Also, please add the package you're using (deb/appimage)

Sveshlex, 2020-02-11:
The problem was in KDE Neon. In kubuntu, the program works correctly.

lephuongbg, 2020-02-27:
Same issue. Can't use it to flash kde neon image in kde neon.

lephuongbg, 2020-08-04:
This is the error output in the dev console. It might be related that KDE Neon use appimagelauncher to integrate AppImage files.

```json
{
  "cause": {
    "killed": false,
    "code": 126,
    "signal": null,
    "cmd": "cd \"/home/lephuong\"; \"/usr/bin/pkexec\" --disable-internal-agent /bin/bash -c \"echo SUDOPROMPT; bash '/tmp/balena-etcher-electron--26632-u4T1lsRBEwQB-.cmd'\""
  },
  "isOperational": true,
  "killed": false,
  "code": 126,
  "signal": null,
  "cmd": "cd \"/home/lephuong\"; \"/usr/bin/pkexec\" --disable-internal-agent /bin/bash -c \"echo SUDOPROMPT; bash '/tmp/balena-etcher-electron--26632-u4T1lsRBEwQB-.cmd'\""
}
```

lephuongbg, 2020-08-04:
I can verify that balena etcher installed using .deb package works fine. The issue is with AppImage package only.