balena-io/etcher #3051: Install conflict (rpm) with other Electron app (draw.io)

skipperTux, 2020-02-03:
- **Etcher version:** 1.5.73
- **Operating system and architecture:** Fedora 31 x86_64
- **Image flashed:** not applicable
- **Do you see any meaningful error information in the DevTools?** not applicable
- References:

#### Steps to Reproduce:
If you try to install `balena-etcher-electron-1.5.73.x86_64.rpm` when [draw.io Desktop](https://github.com/jgraph/drawio-desktop) (rpm) is already installed on your system, you will get the following error message

```bash
Running transaction check
Transaction check succeeded.
Running transaction test
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: Transaction test error:
  file /usr/lib/.build-id/3c/a310025e6d72d037f1fe23888c08317ff5c5f7 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
  file /usr/lib/.build-id/3e/0326bc2ef360cbd8a9d39169b3cffed688d09f from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
  file /usr/lib/.build-id/40/b87885166f5ef16d1873ab3a91c5c8113a7657 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
  file /usr/lib/.build-id/43/403913e1ccc0475b5e2d1d2ddca551b8815a99 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
  file /usr/lib/.build-id/ab/b8ed97d406d9d9dda07e51af4fdf18fb3f5437 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
  file /usr/lib/.build-id/e4/d702c06d6d2db24de306bdf36168f02207268b from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
```

See also cross-reference issue at draw.io Desktop.

#### Current workaround:
Install at least one of those applications as AppImage.

lurch, 2020-02-03:
Related to https://github.com/jordansissel/fpm/issues/1503 ?

maganezf, 2021-06-11:
> * **Etcher version:** 1.5.73
> * **Operating system and architecture:** Fedora 31 x86_64
> * **Image flashed:** not applicable
> * **Do you see any meaningful error information in the DevTools?** not applicable
> * References: [draw.io Desktop Issue 212](https://github.com/jgraph/drawio-desktop/issues/212), [Unwanted directories added to RPM Issue 1503](https://github.com/jordansissel/fpm/issues/1503)
> 
> #### Steps to Reproduce:
> If you try to install `balena-etcher-electron-1.5.73.x86_64.rpm` when [draw.io Desktop](https://github.com/jgraph/drawio-desktop) (rpm) is already installed on your system, you will get the following error message
> 
> ```shell
> Running transaction check
> Transaction check succeeded.
> Running transaction test
> The downloaded packages were saved in cache until the next successful transaction.
> You can remove cached packages by executing 'dnf clean packages'.
> Error: Transaction test error:
>   file /usr/lib/.build-id/3c/a310025e6d72d037f1fe23888c08317ff5c5f7 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
>   file /usr/lib/.build-id/3e/0326bc2ef360cbd8a9d39169b3cffed688d09f from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
>   file /usr/lib/.build-id/40/b87885166f5ef16d1873ab3a91c5c8113a7657 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
>   file /usr/lib/.build-id/43/403913e1ccc0475b5e2d1d2ddca551b8815a99 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
>   file /usr/lib/.build-id/ab/b8ed97d406d9d9dda07e51af4fdf18fb3f5437 from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
>   file /usr/lib/.build-id/e4/d702c06d6d2db24de306bdf36168f02207268b from install of balena-etcher-electron-1.5.73-1.x86_64 conflicts with file from package draw.io-12.5.3-267.x86_64
> ```
> 
> See also cross-reference [Issue 212](https://github.com/jgraph/drawio-desktop/issues/212) at draw.io Desktop.
> 
> #### Current workaround:
> Install at least one of those applications as AppImage.
> 
> Edit:
> 
> * Update reference from draw.io
> * Update reference (thx @lurch)

I have this problem too, on fedora 34. I can only have one electron package installed on my system :(