balena-io/etcher #3052: Is disabling "nodeIntegration" flag planned?

vladimiry, 2020-02-03:
https://github.com/balena-io/etcher/blob/98611267d504d5fb2311553f85810773798b71c9/lib/gui/etcher.ts#L78

thundron, 2020-02-05:
Reason being ...?

vladimiry, 2020-02-05:
General security concerns.

thundron, 2020-02-05:
Just trying to understand - can you be a bit more specific on the use case and the security concerns?

vladimiry, 2020-02-05:
Enabling `nodeIntegration` means direct expose the nodejs functionality to any web page you load in BrowserWindow/BrowserView/WebView/Iframe/etc. So if you load the remote content somewhere in the app there is a security concern of that content to execute dangerous JS code on your system.

I didn't look deep into the code but, from first glance, [this](https://github.com/balena-io/etcher/blob/98611267d504d5fb2311553f85810773798b71c9/lib/gui/app/pages/main/MainPage.tsx#L255) case looks like remote content loading.

You can see more details in https://github.com/electron/electron/blob/master/docs/tutorial/security.md

thundron, 2020-02-06:
I see, thanks for the heads up. Luckily all the remote content is loaded from our websites which are safely monitored, but we'll take a deeper look and see if we can disable it in the next release(s)

thundron, 2020-02-06:
I see, thanks for the heads up. Luckily, all the remote content is loaded from our websites which are secured, and just as the guide says, we use the `<webview>` component (+ no nodejs code is executed from those webviews), but we'll take a deeper look and see if we need to disable the option + enable context isolation

vladimiry, 2020-02-06:
> all the remote content is loaded from our websites which are secured

I have no doubts that the website is secure and monitored. But there might be missed things or things that are not under your control, so there is always a possibility the website to get hacked and malicious code gets injected into the pages. So if there is an option to mitigate the risk it's worth at least review the possibility of applying that option.

> but we'll take a deeper look and see if we need to disable the option + enable context isolation

I'd like to keep the issue open until a deeper look gets taken.

By the way, the simplest way to disable the `nodeIntegration` but keeping using the nodejs stuff from the web content is to move the web logic that requires access to nodejs functionality to a `preload` script and then call the nodejs stuff through the `electron.remote` handle. But the best way is to use pure IPC for all the cross-process communications.

thundron, 2020-02-06:
As I said, we'll take a deeper look and see what can be done