balena-io/etcher #3053: "Continue" does not continue after "Missing partition table" prompt

chdh, 2020-02-03:
- **Etcher version:** 1.5.73 (and 1.5.72)
- **Operating system and architecture:** Windows 7, 64 bit
- **Image flashed:** selfinstall-odroidc2-eng-s905_6.0.1_master-121-v5.6.img
- **Do you see any meaningful error information in the DevTools?**  No

After clicking "Continue" in the "Missing partition table" prompt, the main dialog is still in the "Select image" mode instead of proceeding to "Select target".

This error does not occur with Etcher version 1.5.71. It started with version 1.5.72.

zvin, 2020-02-03:
Thanks @chdh , will fix asap

tristanbuckner, 2020-02-04:
I have the same issue on the mac with version 1.5.73. @zvin is there a download link anywhere for version 1.5.71 to use in the interim?

zvin, 2020-02-04:
This should be fixed in 1.5.74
@tristanbuckner all releases are available here https://github.com/balena-io/etcher/tags

closed 2020-02-04 by zvin

zvin, 2020-02-05:
@chdh fixed in 1.5.75
You might run into another issue though:
if your image has no partition table that etcher recognizes and the `trim` option is enabled (which is now the default), you will get an error when you hit the write button.
Disable the trim option before flashing as a workaround until I fix this.

chdh, 2020-02-05:
@zvin Thanks for the quick fix.