balena-io/etcher #3055: Etcher will not get past gray window at start

mwiegant, 2020-02-04:
- **Etcher version:** 
1.5.74, Type = nsis
- **Operating system and architecture:** 
Windows 10, x64, Intel CPU
- **Image flashed:** 
Did not make it this far
- **Do you see any meaningful error information in the DevTools?** 

> C:\Users\MyUser…js\dist\raven.js:58 {"stack":"TypeError: Cannot convert undefined or null to object\n    at Function.entries (<anonymous>)\n    at Object.init (file:///C:/Users/MyUser/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:3951:49)\n    at main (file:///C:/Users/MyUser/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:3633:12)","message":"Cannot convert undefined or null to object"}
> console. computed	@	C:\Users\MyUser…js\dist\raven.js:58

Currently trying to follow the instructions listed by nvidia, [at this page](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write), for flashing a Jetson Nano. Currently blocked on this process as it's recommended to use Etcher and Etcher will not start. Any suggestions? :)

Trying to find a solution that works, between running in a compatibility mode or a different graphics processor. Would be open to down-grading to an older version too, although I didn't see an option for this on the site.

cweinst1, 2020-02-04:
Im having this same issue. I have been unable to find a solution, and running in compatibility or administrative mode does not seem to help. I am also on Windows 10 x64 with an Intel CPU (NVIDIA GPU) but I have also tried running the installer on my unbuntu virtual machine with no luck. I was unable to find older versions either so may just look for another image writer. Please post if you find any workarounds for the issue with Etcher

mwiegant, 2020-02-04:
@cweinst1 It's not much but I did discover how to download older versions.

https://github.com/balena-io/etcher/releases

Looked at my dev tools when I downloaded, and saw that the download was coming straight from their releases. The file for me (on my windows computer, here) was balenaEtcher-Setup-[version].exe

Also a note, I saw that v1.5.74 was JUST released, 40 minutes ago. Could be a broken build.

mwiegant, 2020-02-04:
Sure enough. v1.5.73 started right up. Seems that the latest version, v1.5.74, is broken at the moment.

cweinst1, 2020-02-04:
@mwiegant Awesome, thanks for providing the link to the releases and checking the old version! Have a good day and good luck with your project

closed 2020-02-04 by mwiegant

lurch, 2020-02-04:
Probably a duplicate of #3057 ?