balena-io/etcher #3057: Black screen when opening Etcher

flightless-katie, 2020-02-04:
- **Etcher version:** 1.5.74
- **Operating system and architecture:** Mac OS Mojave 10.14.6
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** Yes, see below

After tracing the error as the program is running, I found that in leds.ts this code seems to be throwing: (this is on startup, first time trying Etcher)
```
const ledsMapping = settings.get('ledsMapping');
for (const [drivePath, ledsPaths] of Object.entries(ledsMapping)) { // <-- this line throws
  leds.set('/dev/disk/by-path/' + drivePath, new RGBLed(ledsPaths));
}
```
settings.get(...) is returning undefined, so when the for loop tries to use Object.entries(ledsMapping), it is trying to get the entries of undefined, which causes an error.

If in devtools I set ledsMapping to an empty object `{}` the program continues to the opening menu to select image, etc. It's doing a flash operation now with that manual hack. :)

flightless-katie, 2020-02-04:
PS, I am not familiar with your codebase or any conventions you may have, but this could be a potential fix, maybe?
```
const ledsMapping = settings.get('ledsMapping') || {};
```

barrypress, 2020-02-05:
This did work, but not being familiar with the codebase or the dev tools, I had to fumble around a lot before I finally set enough breakpoints (discovering cmd-R was a biggie, too) and pasted in the patch in enough files. I believe I had to do at least gui.js and leds.ts, but not altogether sure which one worked.

However, the fix is not persistent - closing and reopening the pap results in the same gray useless window.

Nightwing531, 2020-02-05:
Same problem here. I would really like this to get fixed

closed 2020-02-05 by zvin