balena-io/etcher #3063: cannot read property UPDATE_NOTIFIER_SLEEP_DAYS of undefined

jelinj8, 2020-02-05:
- **Etcher version:** 
1.4.5 (and several previous)
- **Operating system and architecture:** 
Win10PRO, x64 (but also others)
- **Image flashed:** 
irrelevant

After starting Etcher, usually I get error dialog:

`[Window Title]
Error

[Main Instruction]
Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined

[Content]
TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined
    at Object.exports.notify (file:///C:/Users/jakub/AppData/Local/Programs/etcher/resources/app.asar/generated/gui.js:3834:55)
    at file:///C:/Users/jakub/AppData/Local/Programs/etcher/resources/app.asar/generated/gui.js:3216:27
    at tryCatcher (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:504:31)
    at Promise._settlePromise (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:561:18)
    at Promise._settlePromise0 (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:685:18)
    at Promise._fulfill (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:630:18)
    at Promise._resolveCallback (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:425:57)
    at Promise._settlePromiseFromHandler (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:516:17)
    at Promise._settlePromise (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:561:18)
    at Promise._settlePromise0 (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:685:18)
    at Promise._fulfill (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:630:18)
    at MappingPromiseArray.PromiseArray._resolve (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise_array.js:125:19)
    at MappingPromiseArray._filter (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\map.js:123:10)
    at MappingPromiseArray._promiseFulfilled (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\map.js:95:18)
    at MappingPromiseArray.PromiseArray._iterate (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise_array.js:113:31)
    at MappingPromiseArray.init (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise_array.js:77:10)
    at Promise._settlePromise (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:558:21)
    at Promise._settlePromise0 (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:685:18)
    at Promise._fulfill (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:630:18)
    at Promise._settlePromise (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:574:21)
    at Promise._settlePromise0 (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Users\jakub\AppData\Local\Programs\etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:685:18)
    at Promise._fulfill (C:\Users\jakub\AppDat`

After that, Etcher works normally.

lurch, 2020-02-05:
> Etcher version: 1.4.5

That's a really old version, and the bug you've mentioned has been fixed in newer versions https://www.balena.io/etcher/
(although you probably want to wait until 1.5.75 has been released, as 1.5.74 has known problems)

jelinj8, 2020-02-05:
Sorry, didn't realize that as it proposed an update, which I've installed, but the shortcut was still starting the old version which was not replaced...

closed 2020-02-05 by jelinj8

lurch, 2020-02-05:
Perhaps that was because of the rename from Etcher to BalenaEtcher?