balena-io/etcher #3065: Allow PolKit to identify etcher temporary file

ffissore, 2020-02-06:
Etcher creates a temporary file after hitting the "flash" button. Running that file requires superuser privileges.
However the file name is totally random, which makes it impossible to set up a PolKit rule that allows normal users to run that file without asking them root password
Investigate how to set up such a rule, for example by prefixing the name of the temporary file with a known value

- **Etcher version:** 1.5.51
- **Operating system and architecture:** linux

lurch, 2020-02-06:
Is that what #3047 was attempting to do? :man_shrugging:

ffissore, 2020-02-06:
It looks like so, and it's now part of #3064. Let's see if prefix does the trick, otherwise we'll need to look for something else

lurch, 2020-02-06:
That PR has now been merged, so I guess it's included in Etcher 1.5.76 ? :man_shrugging:

thundron, 2020-02-06:
Correct @lurch , closing

thundron, 2020-03-27:
Did the prefix solve the issue @ffissore ?

ffissore, 2020-03-27:
I'll ping the [forum user](https://forums.balena.io/t/how-to-identify-etcher-with-polkit/58363/28) that requested this change. What's the prefix on linux?

thundron, 2020-04-02:
It's `balena-etcher-electron-`, closing so this gets propagated

closed 2020-04-02 by thundron

georgiats, 2020-05-05:
Not sure if it helps, but the user has found a resolution and reported it here. https://forums.balena.io/t/how-to-identify-etcher-with-polkit/58363/30

thundron, 2020-05-06:
I guess the prefix is working as intended then.
Thank you @georgiats