balena-io/etcher #3067: Non-standard password dialog

adamkdean, 2020-02-13:
Upon making changes, as would be expected, Etcher asks for the password so it can get superuser access but it uses a non-standard dialog. This makes me nervous.

[image]

You'd imagine it to use the standard prompt like #1526.

lurch, 2020-02-13:
AFAIK this is related to #2939

zvin, 2020-02-17:
There is currently no other way for writing to `/dev/rdiskN` on macOS Catalina.
See the linked issues above.

closed 2020-02-17 by zvin

CanonCan, 2020-02-25:
Marcos Catalina broke writing to USB drives for many tools such as the unraid USB creator as wel best to grab a copy of windows and do it in boot camp or a virtual environment

lurch, 2020-05-26:
The reason _why_ it uses a non-standard password prompt is explained in #3107